riscv/riscv-isa-manual #2: Specify PMPs

aswaterman, 2017-02-07:
Current plan is to mirror breakpoint design.  Include feature to protect M-mode from itself, and a lock bit that makes a PMP read-only until next reset.

aswaterman, 2017-02-15:
Re: locking feature: since PMPs are a whitelist, locking them is only useful for security if you lock all of them; else, the unlocked ones serve as an escape hatch.  The PMPs would be more flexible if you could lock some but not others, and still guarantee that certain parts of the address space are inaccessible.

One approach is to create the notion of blacklist and whitelist PMPs, where the blacklist takes precedence over the whitelist.

eternaleye, 2017-02-25:
Another option might be to make the whitelist ordering-sensitive - that is, the first matching PMP for an address wins.

This would have the effect of ensuring that PMPs only take effect on disjoint regions of memory, without requiring any actual overlap checking on the part of the implementation.

M-mode could then mark memory read-only using PMP 0, lock it, and know that no other PMP could override.

aswaterman, 2017-02-25:
Good point, we'll consider that option.

aswaterman, 2017-04-11:
dd8d4755b32db89fcf3771dcfc5a4c3dfa3f6ddf closes this issue

commit 3b13f375 (2021-08-30): Fix conflicting VU-mode Svinval exception text

Closes #2.

Thanks to @jhauser-us
closed 2017-04-11 by aswaterman