riscv/riscv-isa-manual #3: Global trap-enables instead of interrupt-enables

aswaterman, 2017-02-07:
(We haven't decided whether we want to follow through with this proposal.)

Currently, if supervisor takes an exception delegated to supervisor, it always is delegated to supervisor, even recursively. This makes it hard to debug double faults (or handle them specially in the monitor).

The proposal, from Andrew Lutomirski, is to change sstatus.SIE to sstatus.STE (supervisor trap enable). As before, when STE=0, interrupts are disabled; additionally, no exceptions will be delegated down to S. So a recursive exception will go to H-mode or M-mode, even if that exception was delegated to S-mode.

OTOH, it really is possible to do this in software already; some OSes expect the notion of a global interrupt-enable; and it is less desirable for M-mode than for other modes.

aswaterman, 2017-02-08:
If you use the RV Linux discipline where sscratch=0 while in the kernel, it is easy enough to detect double-faults with the current ISA, without using any data memory accesses that might fault:

        trap_entry:
          csrrw sp, sscratch sp
          beqz sp, trap_from_kernel
          # Handle trap from user space.
          
        trap_from_kernel:
          # OK use sp as scratch register here.
          csrr sp, sstatus
          andi sp, sp, SSTATUS_SPIE
          bnez sp, double_fault
          # Handle trap from kernel.
          
        double_fault:
          # Die.
          call sbi_shutdown

aswaterman, 2017-02-08:
So maybe we don't need to change anything?

sorear, 2017-02-09:
What happens in `# Handle trap from user space` ? At that point sscratch contains a non-zero non-percpu value, so you need to spill the old sp somewhere but can't take a trap.

aswaterman, 2017-02-09:
I thought we were trying to deal with kernel stack overflows.  When coming from userspace, you enter the kernel at the top of your thread's stack, so it's safe to write there.

aswaterman, 2017-02-22:
Closed on the basis that the main use case (detecting kernel stack overflows resulting in double faults) can be dealt with in software, as shown above.  Leaving it up to software also gives S-mode software the opportunity to recover from certain double faults.

The other use case (debugging double faults) can be handled with different medeleg/hedeleg settings.  If all exceptions are routed through a more privileged mode, the problematic cases can be detected in software so that more useful diagnostics can be printed.

commit e53dc722 (2021-08-30): Fix missing case in PBMT memory ordering.

Closes #3.

Thanks to @jhauser-us
closed 2017-02-22 by aswaterman