riscv/riscv-isa-manual #6: Vectored interrupts

aswaterman, 2017-02-07:
Not especially important for Unix-like systems, so probably only an optional M-mode feature.

Setting LSB of mtvec could enable vectored interrupt mode.  Not clear we want to simply assign 4 bytes (for a JAL instruction) for each interrupt source.

David-Horner, 2017-02-28:
This extension is possible in v1.11 as the low order bit of mtvec is "enforced" zero currently.

Explicitly reserving the low order bit, (indeed 2 low order bits even) to be zero in the v1.10 is a clean up I'd like to see for all hart writable addresses in CSR registers.  
Reserving the two low order bits allow for future expansion with minimal adverse impact to current designs (it becomes a convention that is beneficial to ensure that C mode cannot be assumed.)

Reserving space allows for non-standard implementations of a vectoring mechanism.

If the attempt is to proved pre-emptive specification so that early implementers do not  dictate a suboptimal defacto standard, then reservation should help address that. As would explicit v1.10 sections that are stipulated as in flux and not meeting the v1.10  "that .. will set the bar for making
backwards-incompatible changes very high indeed" objective.

aswaterman, 2017-02-28:
Agreed.

commit 4dcaa944 (2017-04-07): Add vectored interrupts

Closes #6.
closed 2017-04-07 by aswaterman