riscv/riscv-isa-manual #10: mhcounteren -> mcounteren; mucounteren -> scounteren

aswaterman, 2017-02-09:
Like the interrupt-delegation CSRs, the counter-enable CSRs should belong to the privilege mode above the mode they control.  mcounteren should control whether any lower mode can use the counters; scounteren should control whether user mode can use them.  If a bit is clear in mcounteren, it should appear to be hard-wired to 0 in scounteren.

sorear, 2017-02-09:
"If a bit is clear in mcounteren, it should appear to be hard-wired to 0 in scounteren."

This creates a significant difference in position between S-mode and U-mode.  S-mode code can, by probing scounteren, tell which counters are real and which ones have been disabled.  U-mode code does not have "ucounteren", so it cannot distinguish a disabled counter from a nonexistent counter.

Following this, S-mode can provide emulated counters to U-mode (perhaps to make TIME appear continuous after process migration), but a M-mode stub hypervisor cannot as easily emulate timers to S-mode, because any interference would be visible in scounteren.

aswaterman, 2017-02-09:
Yep.  Strike that last sentence.

commit b38e6f31 (2017-02-21): mhcounteren -> mcounteren; mucounteren -> scounteren

Resolves #10
closed 2017-02-21 by aswaterman