riscv/riscv-isa-manual #9: Reconsider SFENCE.VM/ASIDs

aswaterman, 2017-02-09:
Presently, SFENCE.VM takes its ASID argument from the sptbr.  Since any PTE the sptbr points to can be speculatively loaded into the TLB, you don't want to just modify sptbr.asid arbitrarily.  So, in an OS like Linux, you might flush entry X in ASID Y as follows: (recall ASID 0 is the global ASID, and assume the kernel maintains a page table that consists only of its global mappings)

    tmp <- sptbr
    sptbr <- asid = Y, ppn = global page table
    sfence.vm X
    sptbr <- asid = tmp.asid, ppn = global page table
    sptbr <- tmp

It's not ideal for simple implementations that always flush the TLB on sptbr writes, and it imposes the (minor) constraint that this code sequence is mapped into all address spaces.  And ASID 0 being special is an annoying quirk.

Here is an alternative proposal:

- ASID 0 is no longer special.
- SFENCE.VM takes two arguments: rs1=addr, rs2=asid.
- When rs1=x0 and rs2=x0, it orders prior stores with all subsequent translations.
- When rs1=x0 and rs2!=x0, it orders prior stores to non-Global PTEs with subsequent translations for the specified ASID.
- When rs1!=x0 and rs2=x0, it orders prior stores to PTEs corresponding to virtual address rs1 with all subsequent translations.
- When rs1!=x0 and rs2!=x0, it orders prior stores to non-Global PTEs corresponding to virtual address rs1 with all subsequent translations for the specified ASID.

This requires us to use funct3=4 in the SYSTEM opcode, since funct3=0 has opcode bits within rs2.

aswaterman, 2017-02-15:
Merged into spec.

aswaterman, 2017-02-15:
0b8a8e0c2a4385005e2ef6d92f3b4feee822fd25

closed 2017-02-15 by aswaterman

David-Horner, 2017-02-27:
in sbi.tex a small correction:

line 57 SFENCE.VM to SFENCE.VMA

& \multirow{4}{*}{\parbox{8cm}{Instruct other harts to execute SFENCE.VM. {\tt harts} points to a bitmask of remote hart IDs; NULL indicates all harts. {\tt asid} holds the address-space ID; 0 indicates all address spaces.}} \\

& \multirow{4}{*}{\parbox{8cm}{Instruct other harts to execute SFENCE.VMA. {\tt harts} points to a bitmask of remote hart IDs; NULL indicates all harts. {\tt asid} holds the address-space ID; 0 indicates all address spaces.}} \\

I found a place to note this - so you can disregard the email.