riscv/riscv-isa-manual #13: Interrupt/Exception Priorities

kasanovic, 2017-02-13:
Revisit, and improve explanation of fixed interrupt/exception priority ordering when multiple events are ready to cause a trap.

aswaterman, 2017-02-21:
Interrupts should be ordered by privilege mode, and all interrupts should be higher priority than synchronous exceptions.

David-Horner, 2017-02-27:
Does this imply that all "user-extensions" must either operate synchronously (that is have no asynchronous traps) and if asynchronous behaviour is desired that it must be implemented through an external interrupt?

In part I am thinking about the vector extension that allow extensive pipe-lining, substantially longer than the rest of the core's length. 
Allowing overlapping memory access of later vector elements with the commencement of operations on the initially retrieved set is necessary for optimal performance.
And memory assess issues have high potential for occasional events that require supervisor (in generic sense) management.

Other co-processor implementations would have the same issues (I suspect).
 
Or, is this a separate issue altogether and does not affect what you have here as the stipulated precedence?

aswaterman, 2017-02-27:
Interrupts are asynchronous events and so races between them and with synchronous exceptions are inherent.  I'm concerned about the scenario that interrupts have been disabled for a while, so that multiple interrupts have become pending in MIP.  In that scenario, what transpires once interrupts are re-enabled?

David-Horner, 2017-02-28:
Doesn't PLIC manage this to a significant degree?

A significantly flexible and programmable PLIC can advance or withhold any interrupts (in addition to directing to the various modes that would circumvent a mandated priority order).
 
Further, as interrupts are asynchronous events a race can exist that a notification and intervention by a target is initiated when a higher event is triggered, but cannot practically be aborted. Mandating an order would need to address this kind of preempting which may well be outside the core, and would potentially increase complexity for little value.

If we require it to be stipulated, this sequence of handling should be the canonical order that must be able to be implemented by a conforming implementation, and that is the default order for systems that self initialize the ordering.

Further assessment of the validity of imposing this ordering requires an evaluation of use cases.

I suspect, any active WFI should be given highest priority , whatever mode the machine is in.

Next, if N facility is active it should be given highest priority if the hart is in the correct virtualised user mode.

I doubt these two use cases are trivial to mandate as hart state is the overriding contributing factor.

By the way, should this (and some of the other Issues be on the mailing list(s). And which ones, as these issues are tradeoff issues across ISA/HW (and to a lesser degree SW).

As these

aswaterman, 2017-04-11:
TODO, add commentary about why the interrupt order specified in the MIP/MIE section is the way it is.

r901042004, 2017-07-04:
I have a question here.
If a trap, an interrupt pending with enable and a CSRRW instruction writing into mepc are happening in the same cycle, which action would be handled first? (Assume we only implement M-mode privilege now.)

DSHorner, 2017-07-04:
r901042004:
The result is implementation dependent.
However, it is most likely that the Interrupts should not be enabled when writing to mepc, as the interrupt handling will destroy the mepc value. A subsequent instruction - most likely a mret - will not be able to depends on that value. Similarly, one needs to ensure that the code cannot case a synchronous exception between the CSRRW to mepc and its use (again likely a mret).

r901042004, 2017-07-04:
DSHorner:
Thank you. I got it.

closed 2018-01-24 by kasanovic

gxflying, 2020-08-28:
@aswaterman 
Hi , I am confused about this question too .
the spec says "the exceptions have lowwer priorities than interrupt to minimize worst-case interrupt latency"
BUT  how can this go as what we expect when exceptions can NOT be disabled?  
**when in interrupt handler , exception may happen at any time which will definitely delay the interrupt handling** 
or this priority here is just  all about interrupt & exception accuring in the same cycle ,which means if they occur in the same cycle , the code should go to interrupt handler other than exception handler?
above all, I cann't understand how to "minimize worst-case interrupt latency" by setting "the exceptions have lowwer priorities than interrupt"  while exceptions can NOT be disabled ...
do I misunderstand  or miss some points ?
could you please help to clarify this?
thank you~

https://github.com/riscv/riscv-isa-manual/blob/bb5537941b2bc31acbf111d9261ee6da1aae4a50/src/machine.tex#L1590

sorear, 2020-08-28:
In many cases the trap handler will be privileged code that can be trusted to not generate exceptions, while the code being interrupted is untrusted.

commit def947b7 (2018-01-24): Added commentary on fixed interrupt priority scheme for mip/mie.

Closed #13