riscv/riscv-isa-manual #15: Add commentary explaining removal of Mbb

aswaterman, 2017-02-19:
Between PMPs and PIC, many use cases are covered.

Worth mentioning that mstatus space is reserved for that purpose.

David-Horner, 2017-02-28:
The load, store    can specify x0 as the base register for the address, and therefore access the high and low 4 K of the memory map. 
As can their compressed variants, but with a narrower range.
ditto for  JALR (but not C.JALR) instruction.

Notably, this usage of JALR is called out in the specification as a candidate for the system call invocation.

Mbb allowed to map these addresses for these instructions to a useful memory range (out of ROM, for instance which might be the machine configuration default).

And Mbbid further allowed having separate spaces for load/store vs JALR.

Can PMP as currently defined/envisioned remap zones?
For example address bit inversion?
And can it differentiate between data and instruction fetch (to effectively implement Harvard Arch?)
 
If possible, can the explanation of the Mbb modes address these?

Thanks.

aswaterman, 2017-02-28:
PMPs don't remap, so the low 2 KiB is of limited use in this proposal--or at least of different use, e.g., shared code or constant pools.

It's not that we think Mbb is a bad idea, just that it's not a high priority right now.  It could rematerialize in a compatible manner in a future version of the spec.  (This all belongs in the rationale.)

kasanovic, 2017-03-21:
Added text to preface.

closed 2017-03-21 by kasanovic