riscv/riscv-isa-manual #16: Document `N' extension

aswaterman, 2017-02-23:


sorear, 2017-04-05:
Our existing WARL mechanism allows `RV32IS` implementation to hardware `spbtr` to 0 and `mstatus.SPP` to `S` and let the allowed values of `mstatus.MPP` be S and M.  Up to a relabeling of control registers this is equivalent to `RV32INU`, so the latter seems a bit redundant.

Having `N` as a distinct extension allows `RV32INSU` to be considered.  For that I will note that, since `uepc` is writable user-mode state that does not participate in the `sstatus.XS` protocol, `N` as has been considered so far is incompatible with supervisor code that does not expect it.

aswaterman, 2017-04-11:
My preference is still to ditch the N extension altogether and use S-mode with hard-wired ```sptbr``` for these systems.

The N extension is indeed incompatible with supervisors that don't expect it.  But this is true of any new extension with state accessible to U-mode.  So, the SBI must guarantee that any such extension is initially disabled.

aswaterman, 2018-11-21:
Note that https://github.com/riscv/riscv-isa-manual/issues/135 must be addressed when defining N--that is, either WFI must be disallowed, or a mechanism to trap WFI must be added.

aswaterman, 2018-12-21:
Closed by 87a1e5f400f989467831a9352265c82b4a1848aa

closed 2018-12-21 by aswaterman