riscv/riscv-isa-manual #17: Make WFI M-mode-only, or add an mstatus bit to trap WFI

aswaterman, 2017-02-24:


David-Horner, 2017-03-04:
>The TW (Timeout Wait) bit supports intercepting the WFI instruction (see Section 3.2.2). When
TW=0, the WFI instruction is permitted in S-mode.

It appears to be here now, at least in this version and rendering of the function.
[https://github.com/riscv/riscv-isa-manual/blob/master/src/machine.tex#L621-L627](url)

I'd suggest that other instructions also be intercepted in the same way.
Specifically, jump and taken branch with zero offset.

> the WFI instruction causes an illegal instruction trap

I'm concerned about the overloading of **illegal*** instruction trap:
1. Some of the trapped instructions are truly illegal in a certain context.
2. Others are unimplemented instructions.
3, And others such as this are optimization traps.

I would suggest "decoder exception" be used as the cause meaning for 1 and 2.
Perhaps this timeout should be a different mcause.
Or, as it is potentially asynchronous in nature, not determined by issues in execution, perhaps it should be an interrupt to M-mode.

I would like this suggestion opened as a separate issue for discussion.
Or should I raise it on the ISA group?

aswaterman, 2017-03-04:
Overloading the same trap cause is the intent.  Perhaps the name "illegal instruction" should change.  John Hauser has suggested "invalid instruction" instead (for different reasons).  I don't like "decoder exception" because it sounds microarchitectural.

David-Horner, 2017-03-04:
I expect differentiating causes will be valuable for efficient trap handling.
Some implementations will want to provide more specific "cause" feedback.
I believe microarchitectural aspects, e.g. where in the handling of the instruction execution the trap occurred to be a logical and effective distinction, and so "decoder exception" would be one such.
Looking at the list:

> Instruction address misaligned
> Instruction access fault
> Illegal instruction
> Breakpoint
> Load address misaligned
> Load access fault
> Store/AMO address misaligned
> Store/AMO access fault

don't  they all have a comparable microarchitectural aspect?

For example, the "access fault" could encompass "misaligned", but are differentiated in the codes.
And "Breakpoint" is separated from "illegal instruction" although a valid implementation could reasonably use the latter and determine EBREAK/C.EBREAK from the opcode.

I agree that tailoring to a specific microarchitecture is to be avoided, but the decode function is inherent and performed early in most of our target implementations. Even an alternative implementation, that executes in parallel anticipated instructions and selects the correct results based on the instruction would be able to benefit from this decode distinction.

I believe we have to address the extent to which causes codes are overloaded.
This discussion is complimentary to the vectored trap, deferred to V.11. But unlike the details of implementing that facility, the decision on what the vector values can be to maximize its effectiveness is a V.10 concern.

Do we have any criteria that determines the granularity of the "cause codes"?
It appears the decisions are not related to necessity, as pointed out above, but rather to perceived/anticipated required efficiency, and this implies some microarchitectural bias .
A minimal implementation that performs substantial instruction emulation would benefit from further granularity, as would higher end that utilize the proposed vectored trap mechanism.
I see it as a win-win, that is compatible with opt-out for the further codes, just as Breakpoint can be.

aswaterman, 2017-03-13:
Closing this, since TW has been added.

The other discussion may be reopened as another thread, but I will comment that I don't see much reason to revisit it.  The main reason for the granularity of cause codes is the exception-delegation mechanism.  Since the point of these trap-on-X instructions is to transfer control away from S-mode, overlaying them on the illegal instruction trap makes sense, since that exception should never be delegated.  (Our implementations don't even support delegating it.)

closed 2017-03-13 by aswaterman