riscv/riscv-isa-manual #18: sstatus/mstatus SPV bit

aswaterman, 2017-02-27:
Even though we won't spec the Extended Virtualization scheme just yet, we should add one tiny piece of it to mstatus/sstatus now.  This bit is necessary to emulate Extended Virtualization on systems that don't have it:

SPV ("Supervisor Previous Virtualization mode")

    If S mode is not supported, this bit is a read-only zero.

    Bit SPV affects the behavior of an SRET instruction.  If extended
    virtualization is not supported, and SPV = 1, SRET causes an invalid
    instruction trap.  If extended virtualization is supported, the
    effect of SRET when SPV = 1 is not yet documented.

    (Bit SPV is required to exist even when extended virtualization is
    not supported so that the behavior of extended virtualization may be
    emulated through software if desired.)

sorear, 2017-02-27:
Per our discussion this may not be needed; EV can a separate instruction (SRETV, HRET, call it something else) to enter guests; KVM at least does not try to unify the guest-entry and return-to-user paths.

aswaterman, 2017-03-09:
The problem is that this discussion extends to M-mode, too.  The options there are for the "MPV" bit to modify MRET's behavior, or for a new instruction, MRETV, to be introduced.  The same branching issue arises there: the exit path must use the correct instruction.  Unlike in KVM, the M-mode trap handlers are quite unlikely to use different exit paths, so the MRETV approach requires another control register read and branch.  Seems cleaner for MRET to just do the "right" thing.

Then, for consistency between the privilege modes, having just one SRET whose behavior is affected by SPV seems like the best thing to do.

sorear, 2017-03-09:
@aswaterman it wouldn't be M *P* V because the supervisor is still virtualized while handling a M-mode trap, and the relevant state would be live in sstatus.TSI and m2atp etc

aswaterman, 2017-03-09:
Wherever the state lives, the point still stands.

aswaterman, 2017-03-13:
A better solution is spec'd in 8bb62be.

closed 2017-03-13 by aswaterman