riscv/riscv-isa-manual #19: MXR for PMPs, not just paging?

aswaterman, 2017-02-27:
MXR is currently defined to affect only paging.  I think we need to define MXR to affect PMPs, not just paging.  If a PMP marks a region is U-mode execute-only, it is currently impossible to use the MPRV mechanism alone to emulate that instruction.

aswaterman, 2017-02-27:
Note this would make M-mode X-only PMPs effectively useless (except on M-only implementations, where the MXR bit does not exist).

sorear, 2017-02-27:
My thought earlier was that MXR would be honored only when MPRV was in effect (ideally, R-only pages would fault when accessed under MXR+MPRV)

aswaterman, 2017-02-27:
That makes it less likely for software to make MXR-related bugs, but doesn't close the M-mode X-only hole (as you can set MPRV=1, MXR=1, MPP=M).

aswaterman, 2017-03-02:
Plan is to leave it as defined (only affects paging, not PMPs), and see what happens.

closed 2017-03-02 by aswaterman