riscv/riscv-isa-manual #24: rename *ptbr to *atp?

kasanovic, 2017-03-03:
problem: *ptbr has "r" on end for register, which is not convention (except legacy fcsr), also contains more than a page table pointer, and not all uses will include a page table pointer.
proposal: *atp for "address translation and protection"

aswaterman, 2017-03-03:
I like *atp better, but binutils is released and has sptbr name.  Would be best not to change at this point.

asb, 2017-03-03:
You could always add *atp and keep sptbr as an alias for compatibility.

aswaterman, 2017-03-03:
That's true.

I'm fine either way.

David-Horner, 2017-03-04:
I'm hoping that upstream implementations do not entrench obsolete and misleading conventions/specifications, but that they can be depreciated once better alternatives are provided.

aswaterman, 2017-03-10:
Hauser's take:

> While there are a lot of plausible options, I see the register as
> primarily specifying the root page table:  its location in memory, how to
> interpret it (MODE), and what ASID to associate with it.  I like the word
> "root" better than "base" here, because the active page tables obviously
> form a tree, and "root" is the standard term for this node in a tree data
> structure.  With that logic, I'd probably choose the name 'srpt'.

kasanovic, 2017-03-21:
Not all values put in this register will necessarily be pointers to a memory-resident table (e.g., some future base+bound scheme that is specified entirely by bits in the register), hence why I was suggesting "*atp" as generic term for address translation and protection specifier. Can be usefully misinterpreted as address translation pointer by reader.

commit 26f0d7a5 (2017-04-12): SPTBR -> SATP

Closes #24.
closed 2017-04-12 by aswaterman