riscv/riscv-isa-manual #25: Add rationale for choice of divide-by-zero choice

kasanovic, 2017-03-03:
Shouldn't be undefined.
-1 was value seen in other simple implementations and drops out of simple hardware implementations.

kasanovic, 2017-04-02:
Added rationale for choice of overflow quotient values to spec.

closed 2017-04-02 by kasanovic

karlosz, 2018-01-08:
I think there is a typo in the rationale where it says `and this check can be inserted after the divide`. The branch to check for division by zero should not be done after the divide instruction. At that point the divisor may have been destroyed, ie. if it was the destination of the divide instruction. Instead, the check must be inserted before the divide instruction.

aswaterman, 2018-01-08:
That was intentional; putting the branch after the divide is desirable for code scheduling.

The compiler understands that the destination will be overwritten even on a divide by zero, and can choose the destination register accordingly.