riscv/riscv-isa-manual #32: Replace RISC-V specific Vendor ID in mvendorid with JEDEC manufacturer ID

kasanovic, 2017-03-20:
This would avoid duplication of effort, given that chip makers already need/have a JEDEC code.  It would also free Foundation from managing the vendor ID list.  Registering a JEDEC code costs $500, and can be used for all chips from a manufacturer. mvendorid can always return 0 if no code available.

kasanovic, 2017-03-20:
For soft cores, the vendor ID would be that of the softcore provider, who I guess might not otherwise need a JEDEC code, though this is still useful for debug port interrogation.

kasanovic, 2017-03-20:
Updated spec.

asb, 2017-03-20:
The specification should really indicate the mapping of JEDEC ids to a 32-bit register. JEDEC's encoding is very inefficient (assigned in "banks" of 7-bits, with a "continuation code"). They are beyond 9 banks now, so JEDEC's encoding wouldn't fit in a 64-bit register. Specifying the continuation bits be concatenated with the manufacturer code would be the logical encoding.

aswaterman, 2017-03-20:
Good point @asb; I believe the debug spec already writes up how it maps to 32 bits, so we can crib that text.

aswaterman, 2017-03-20:
mvendorid[6:0] = JEDEC[6:0]
    mvendorid[XLEN-1:7] = number of 0x7f continuation codes

asb, 2017-03-20:
Fixed by 2447f7421b0f4

closed 2017-03-20 by asb

kasanovic, 2017-03-20:
This representation is standardized in section 11.2.1 (Manufacturer Identity code) of JTAG 1149.1 standard and is what I intended.