riscv/riscv-isa-manual #35: Classify and advertise optional feature under standard extensions

Ilanpardo, 2017-03-28:
There are several features in the spec which are optional ( or so I understand that). Some are visible to user mode although I refer here in general for privilege modes as well.  For example the performance monitor counters are optional, also the amount is optional and there are many of those so might be considered expensive with some implementations. Now we have standard extensions for user mode feature groups and there is an enumeration scheme in misa register to tell SW what is supported and what not. I suggest:
1. classify optional features to "extensions" e.g. perfmon extension
2. For features related to SW development (perfmon, debug) there may be benefit to allow more flexibility than with functional features (such as Floating Point). For example an implementation may support perfmon with limited amount of counters. So there should be a way to tell SW how much of such extension is actually supported.
3. Suggest to allow user mode to read the feature advertisement (misa)

kasanovic, 2017-03-28:
Standard RISC-V protocol is to use WARL registers to allow software to directly interrogate platform for existence/number of a given optional feature wherever possible.  In addition, we will be defining "profiles" to give minimum feature set software can expect from a given platform to avoid having to cope with arbitrary combination of parameters.

Ilanpardo, 2017-03-29:
- In section 3.1.16 Hardware performance monitor section, mhpmevent# are defined as WARL as you say. 
- In section 2.2 CSR Listing these are defined as MRW. There is no  uhpmevent#.
Do you assume user code gets this via ABI call? In this case, the ABI will tell you what is the event meaning as that is platform specific, right? something like string value with null representing nothing counted?
 
Same question for misa - Do you assume user mode gets info about supported extensions only via ABI?
Compiler need to generate emulation code for unsupported extensions (e.g. FP).

aswaterman, 2017-04-12:
Yes, all of these will be the responsibility of the ABIs.  Linux already provides a mechanism to find the ISA string: you can read /proc/cpuinfo.

The ABI will also have to define a way to set up the event counters (and in turn the SBI will have to provide a way for the supervisor to set them up, since the event selectors are M-mode registers).

closed 2017-04-12 by aswaterman