riscv/riscv-isa-manual #40: Clarify that virtual and physical address space are circular

sorear, 2017-04-04:
There's been some confusion about this on the list in the past, but I think we intend that:

* load/store address generation overflows silently
* PC incrementing overflows silently
* `*(uint32_t*)(uintptr_t)-1` is treated like any other page-spanning access, and will (possibly after fixup) generate accesses to the first and last pages of memory if both are accessible
* the same rules apply in M-mode or with paging disabled, although in practice the circle is likely to be broken by a PMA or PMP boundary

aswaterman, 2017-04-04:
Agreed on all counts.  Not sure where best to clarify this in the spec.

kasanovic, 2017-04-11:
We probably need to put together a single section/chapter that covers all the various aspects of the memory model, including this issue, misaligned loads, and the PMP fragmentation issue, as well as general memory model.

aswaterman, 2018-11-30:
We should add a statement to this effect in the new Memory section in the unprivileged Intro.

aswaterman, 2018-11-30:
Commentary should also mention that most implementations won't have the same PMAs at -1 and at 0, so the access will fail anyway.

commit 9674e5a7 (2018-12-21): Address space is circular

Closes #40
closed 2018-12-21 by aswaterman