riscv/riscv-isa-manual #41: Document device tree binding

sorear, 2017-04-04:
We need something somewhere that defines the allowed properties for RISC-V harts, PLICs, and any other node types we create.

Another issue is that some resources are potentially shared between M-mode and S-mode; e.g. a UART could be documented in the device tree, but also used to implement `SBI_CONSOLE_PUTCHAR`; S-mode payloads need to know this and not try to use both at the same time.  There's probably an established solution to this somewhere.

aswaterman, 2017-04-12:
I think this should be a separate document from the privileged spec (even including things like the ```SBI_CONSOLE_PUTCHAR``` interaction).

neuschaefer, 2017-05-29:
Please consider asking the (linux) devicetree mailing list (devicetree@vger.kernel.org) for design guidance. Usually, all new devicetree bindings that enter linux's collection of bindings have to undergo review on that list.

palmer-dabbelt, 2018-06-06:
We're created a whole separate document for this one as well: https://github.com/riscv/riscv-device-tree-doc .

closed 2018-06-06 by palmer-dabbelt