riscv/riscv-isa-manual #42: Locking idioms are incomplete

sorear, 2017-04-04:
The user spec provides this guidance:

> We recommend the use of the AMO Swap idiom shown above for both lock acquire and release
to simplify the implementation of speculative lock elision [25].

I've been puzzling over this for months and I don't currently think it's possible to implement `pthread_mutex_lock` using a single `amoswap.w.aq` and `amoswap.w.rl` in the uncontended path while supporting existing `futex`/`WaitOnAddress` APIs, `PTHREAD_PROCESS_SHARED`, and fairness.

(In particular, if a mutex is released with a single AMOSWAP, then it goes directly from "contended" to "unlocked", which allows another process to acquire the mutex before the wait list can be processed.)

aswaterman, 2017-04-04:
(Just providing context:) glibc's low-level locking code seems to use CAS to acquire locks and swap to release them.

aswaterman, 2017-04-12:
I think all there is to say in the spec is that some more complex locks will be implemented with LR/SC.  The only interesting thing about those sequences is how to set the AQ/RL bits on the two operations; while I think it will suffice to set AQ on a lock-acquiring SC and RL on a lock-releasing SC, we'd rather wait til the memory model discussion is more complete before writing that up.

arkflpc, 2017-05-24:
I think that lock elision isn't compatible with LR/SC mechanism. Elided store wouldn't invalidate reservations held by other harts after LC instruction and SC would succeed. It would made LR/SC suffer from ABA problem.

sorear, 2017-05-24:
There are a lot of misconceptions about ABA problem.  LR/SC only has an expressiveness advantage over CAS if unrelated memory accesses are allowed to occur between the LR and the SC; RISC-V forbids this so RISC-V LR/SC is already precisely equivalent to CAS..

kasanovic, 2018-11-07:
Regarding ABA - in the case where there is a single word being accessed, then LR/SC will abort if a second hart rewrites the same value as is currently in memory to the single word under reservation by a first hart.   CAS will not detect this. This is admittedly a limited case but does mean LR/SC is not equivalent to CAS for ABA.

closed 2018-11-07 by kasanovic