riscv/riscv-isa-manual #43: Guidance requested: instruction for spin-wait loops

sorear, 2017-04-04:
Linux is presently using `div a5,a5,zero` for `cpu_relax` (actually the register is variable but it winds up being `a5` in 80% of sites); glibc and Go currently generate no instructions for `atomic_spin_nop` and `runtime.procyield`.

aswaterman, 2017-04-04:
```div a5,a5,0``` is a Rocket-specific hack and surely won't generalize.  (It's actually probably really bad for OoO microarchitectures.)  I'd leave it empty in Go until one of the ```HINT```s is reserved for this purpose.  I'd suggest ```addi x0, x0, 1``` a.k.a. ```c.li x0, 1``` for this purpose.  @kasanovic is this something we want to do?

aswaterman, 2017-04-12:
We're now thinking that FENCE w,nothing (i.e. PW=1, other bits=0) is the right idiom for this, as I believe someone suggested in another thread.

Part of it is that we think other FENCE xxxx,nothing or FENCE nothing,xxxx patterns will be useful as other memory and I/O related hints, but we don't need to get into that now.

Using a compressible instruction doesn't seem especially important in this case (e.g. the divide idiom shows up only 117 times in my vmlinux).  Also, we'd like to preserve the ADDI hint space for something that might actually use the rs1+immediate value the instruction naturally computes, e.g., address-related hints.

@kasanovic will write this up.

aswaterman, 2019-06-24:
Now that the base ISA is ratified, it seems like a good time to address this.  I've made the PR #398.

PR #398 (2021-02-11): Add PAUSE hint
Closes #43
closed 2021-02-11 by aswaterman