riscv/riscv-isa-manual #44: Inconsistency on the Stack Pointer Alignment Specification

hirooih, 2017-04-07:
I've been studying "The RISC-V Instruction Set Manual Version 2.1", and find an inconsistency.
It is in Chapter 20 Calling Convention.  It seems that the chapter is not included in this repository, but I could not find any other places. Let me know if there is a better place.

On P.109 the first paragraph says;

> ... the stack pointer is always kept 16-byte aligned.

On the other hand "20.3 Soft-Float Calling Convention" says;

> ... and the stack discipline is the same except that the stack is only kept aligned to XLEN/8-byte boundaries ...

And the footnote says; 

> The reduced stack alignment saves space in the memory-constrained systems that
might commonly use soft floating-point.

The first paragraph on P.109 should be;

> ... the stack pointer is always kept 16-byte aligned **_unless the soft-float calling convention is not used (see 20.3)_**.

This is a trivial issue, but a real problem is in the definition of C.ADDI16SP instruction, which is also for memory-contrained systems.
It assumes the stack pointer is kept 16-byte aligned and is useless if the stack pointer is not 16-byte aligned.

There should be some description about this restriction in the description of the instruction on page 86.  

And the footnote on the page;

> In the standard RISC-V calling convention, the stack pointer sp is always 16-byte aligned.

also should be fixed.

sorear, 2017-04-07:
The calling convention documentation will be removed from the ISA manual 2.2 because it is being moved to https://github.com/riscv/riscv-elf-psabi-doc/blob/master/riscv-elf.md, although I see that document is silent on stack alignment.

aswaterman, 2017-04-11:
Closing, as this should be discussed on https://github.com/riscv/riscv-elf-psabi-doc/issues/21

closed 2017-04-11 by aswaterman