riscv/riscv-isa-manual #45: Feedback Comments for C spec

hirooih, 2017-04-07:
P.86 Integer Register-Immediate Operations

First it was not clear for me why dest==0 is allowed C.ADDI,  even though it is not allow for C.ADDIW.
If I understand correctly, it is for C.NOP which is encoded as c.addi x0, 0.   I think it is better to have some description about it here.

CS format instructions on P.88

These 6 arithmetic instructions, C.AND, C.OR, ... C.SUBW, are categorized in the CS (Store) format.
But the format is different from any other formats (See Table14.1).
I think we should give them a unique format name, fox example CA (Arithmetic or ALU) format.

kasanovic, 2018-11-07:
rd=0 is not allowed for either C.ADDI or C.ADDIW.  C.NOP is treated as a separate encoding in table, though implementations can implement as C.ADDI with destination of 0.

Agree on CA format - will be added.

closed 2018-11-07 by aswaterman

hirooih, 2018-11-07:
Hi,

I've been completely forgetting about this ticket.

I referred v.2.1 when I wrote this.  I've checked v2.2 but it remains same.

v2.2 P. 76 Integer Register-Immediate Operations

>  The rd/rs1 field C.ADDI is described as **"dest", not "dest !=0"**.

P.79 NOP Instruction

>   C.NOP is encoded as c.addi x0, 0 and so expands to addi x0, x0, 0.

P.82 Table12.5

> Both C.NOP and C.ADDI are descrbied.  The rd/rs1 field C.ADDI is described as **"dest != 0", not "dest"**.

P. 76 and P. 79 are consistent.  But they are not consistent with P.82 a little bit.

aswaterman, 2018-11-07:
If this is a new issue, please open a new ticket

hirooih, 2018-11-08:
> If this is a new issue, please open a new ticket

No. It is an old issue opened on April, 2017.
Please take your time to read and reopen it.

hirooih, 2019-03-10:
Just a note.
I've confirmed all issues I reported are fixed on riscv-spec-imfdqc-ratification-20190305.pdf.
Thank you.

commit 00557c31 (2018-11-07): Define new RVC format CA; state that C.AND, etc. use it

This is not a functional change, just an improvement to the description.

Resolves #45