riscv/riscv-isa-manual #48: SXL/UXL vs. suisa/msisa

aswaterman, 2017-04-11:


sorear, 2017-04-12:
Has something come up that motivates per-thread extension control?

Advantage of limiting to XLEN and keeping it in mstatus is fewer things to swap, for supervisors and hypervisors.  Per-thread extension control is a novel complication and I'm interested to know more about it.

eternaleye, 2017-04-12:
One thing I wonder is if that might provide a "generic" mechanism for delayed state-saving - set `suisa` to a conservative value on context switch, trap on state-accessing operations, save/load state, flip the appropriate bit, and return from the trap.

sorear, 2017-04-12:
That'd be one use, although I don't personally see it as justifying by itself.  (I still need to refile the XS issue)

aswaterman, 2017-04-12:
We're not taking a position on the matter yet, just noting we need to resolve it at some point.  We may decide to punt this 'til priv-1.11, as it won't introduce a backwards incompatibility.

@sorear this decision may affect context-switch cost for some hypervisor designs, and we'll factor that into our proposal.

@eternaleye I'm hoping that, for the English-lettered ISA extensions, we won't feel the need to significantly extend the architectural state beyond the FP and vector extensions.  In particular, the V extension's design allows new extensions to cleanly reuse the same state by reconfiguring the machine.  But your point's taken.

aswaterman, 2017-05-05:
We plan to proceed with the UXL/SXL approach and punt to M-mode for control of ISA extensions.  We leave the door open to to the future definition of *isa CSRs.

closed 2017-05-05 by aswaterman