riscv/riscv-isa-manual #49: mepc, sepc, mtval, stval should be WARL values

aswaterman, 2017-04-12:
They don't need to be a full XLEN bits.

mepc and sepc only need to hold physical/virtual addresses.  To distinguish valid addresses from invalid ones, one additional bit may be necessary--e.g., for Sv39, a 40-bit sepc is necessary to distinguish that bits 63:39 weren't copies of bit 38.

mtval and stval need to be at least as wide, but may be also be wider to support holding 48/64-bit instructions, if that feature is implemented.

This is kind of awkward to spec, but it's a lot of flops, so it's worth saving them.

sorear, 2017-04-12:
Alternately we could handle noncanonical JAL(R) like misaligned JAL(R) and trap at the jump instead of the target?

aswaterman, 2017-04-12:
Straight-line code can cause this problem, too (e.g., if the instruction at address ```0x3ffffffffe``` in an Sv39 system isn't a jump), so I think the extra bit is the way to go.

commit 8fd6ceb2 (2017-04-18): mepc, sepc, mtval, and stval are WARL

Closes #49.
closed 2017-04-18 by aswaterman