riscv/riscv-isa-manual #50: Sv48 cannot be levelled

sorear, 2017-04-13:
VM migration across heterogeneous machines is typically done by preventing VMs that need to be migrated from seeing any feature which is not available in the entire migration pool (e.g. "minimal CPUID" in https://github.com/riscv/riscv-isa-manual/issues/30#issuecomment-288154973).  In general, this works for features where support is explicitly indicated, and does not work for features where support is probed.  

Fortunately, most WARL fields in the privileged architecture are in M-mode CSRs and M-mode is not intended for nonclassical virtualization.  There are however two exceptional S-mode WARL fields which could cause migration trouble:

1. MODE in SATP: if the cluster contains some Sv39 hw and some Sv48 hw, there is no way to force OSes to use Sv39 in VMs that will be migrated
2. UXL in sstatus

aswaterman, 2017-04-13:
I think we can just rely on mstatus.TVM to intercept SATP accesses for the unusual case of migration pools with mixed-Sv39 and Sv48 hardware.

aswaterman, 2017-05-05:
More generally, the answer is that the SBI should provide a mechanism to tell the OS which features are actually supported.  Then, the OS should not dynamically probe those features.

closed 2017-05-05 by aswaterman