riscv/riscv-isa-manual #56: does rem*w sign extend result on divide by zero?

kasanovic, 2017-05-03:
From John Chen email on Jan 6, 2017:
"REMW and REMUW instructions are only valid for RV64, and provide the corresponding signed and unsigned remainder operations respectively. Both REMW and REMUW sign-extend the 32-bit result to 64 bits."
and for divide by zero 
"The remainder of division by zero equals the dividend."
It seem the above description is conflicted .

aswaterman, 2017-05-03:
The result is sign-extended.

Given the context of the previous sentence describing DIVW/DIVUW, it seems clear to me that "dividend" means rs1[31:0] and "remainder" means rd[31:0], so these statements don't seem to conflict.

kasanovic, 2017-05-03:
It's not crystal clear in text, I'll try and be more explicit.

kasanovic, 2017-05-03:
Clarified this in text.

closed 2017-05-03 by kasanovic

aswaterman, 2017-05-03:
looks good, thanks