riscv/riscv-isa-manual #58: Clarify stval for misaligned accesses which cause page faults

sorear, 2017-05-11:
Suppose page 0x80 is valid, page 0x81 is invalid, and the following is executed:

```
lui a0, 0x81
lw a0, -1(a0)
```

Does `stval` get written with the access address `0x80fff`, the start address of the faulting page `0x81000`, or an unspecified faulting address within the access `0x81000 - 0x81002` ?  riscvemu, qemu, and bbl all do the third option, and I don't think anything else makes sense, but it should be specified.

( @s-macke brought this to my attention )

aswaterman, 2017-05-11:
Not the first option.

I think we forgot to specify this (it is arguably a platform spec issue) but I favor the second option for determinism.

sorear, 2017-05-11:
If a core does cross-page misaligned access in hardware then it probably needs to set `stval` in hardware, which I think makes this not-a-platform-spec issue?

aswaterman, 2017-05-12:
Fair point.  "stval is written with the faulting address" already rules out option 1, as the fault was caused by one of the addresses after the page boundary.  But as written, it's currently left to the implementation.

@kasanovic and I have discussed requiring that misaligned accesses be emulated with a sequence of single-byte accesses, ascending from the base address; or perhaps instead with two accesses of the original size to ```base & -size``` then ```(base + size) & -size```.  The latter has the advantage of matching the obvious hardware implementation of misaligned accesses.  However, the former has the advantage of not triggering false-positive PMP faults.

In all cases (including hardware implementation of misaligned accesses), it seems natural to mandate that the fault occur on ```0x81000```.

sorear, 2017-05-12:
I assume this is for the purposes of the memory model?  Remember that bbl can generate more than 4 accesses to emulate a `sw` if the access faults and needs to be restarted. :/

`0x81000` is fine by me, but all we really need now is "stval points to the faulting page, not the one before it."

aswaterman, 2017-05-12:
Yeah.

Agreed, "stval points to the faulting page" is sufficient for practical purposes.

aswaterman, 2017-09-22:
Partially clarified in e2ae148dcca7f725a577ca911059edb337b1ef25, but I didn't add language about the lowest-numbered byte.

aswaterman, 2018-11-10:
I think this is actually "resolved enough," but feel free to reopen a new issue if you disagree.

closed 2018-11-10 by aswaterman