riscv/riscv-isa-manual #59: sign extention of shamt for C.SLLI/C.SRLI/C.SRAI in RV128

DSHorner, 2017-05-12:
In both V2.1 and V2.2 the C.SRLI and C.SRAI shamt are sign extended in RV128.

> Furthermore, the shift amount is sign-extended for RV128C, and so the legal shift amounts are 1–31, 64, and 96–127. C.SRLI expands into srli rd 0 , rd 0 , shamt[5:0], except for RV128C with shamt=0, which expands to srli rd 0 , rd 0 , 64

Please confirm that the sign extension is only to the rightmost 7 bits of the immediate in the RV128 SRLI and SRAI.

C.SLLI does not mention the sign extension, but only the 0 as 64 for RV128.

> For RV128C, a shift amount of zero is used to encode a shift of 64. C.SLLI expands into slli rd, rd, shamt[5:0], except for RV128C with shamt=0, which expands to slli rd, rd, 64.

Please confirm that shamt for RV128 C.SLLI is intended to function as for RV128 C.SRLI/C.SRAI.

(Clifford's query about C.ADDI and rd/rs1 == 0 got me looking at the compressed instructions again)

sorear, 2017-05-12:
The comment on the top of page 78 of https://cdn.rawgit.com/riscv/riscv-isa-manual/master/release/riscv-spec-v2.2.pdf#page=90 indicates that the range of right shifts and left shifts is deliberately different on RV128 and gives two reasons why.

DSHorner, 2017-05-12:
And the explanation was also in V2.1 - sorry for the noise.

closed 2017-05-30 by DSHorner