riscv/riscv-isa-manual #65: Propose adjusting FMIN/FMAX NaN handling for better standard conformance

jhauser-ucberkeley, 2017-05-19:
The current definitions of instructions FMIN and FMAX do not match the behaviors defined for any of the **minimum** and **maximum** operations in the current working draft of the next revision of the IEEE Standard for Floating-Point Arithmetic (IEEE 754).  This is a change expected in the the IEEE Standard compared to the previous version.  The differences concern the results returned for NaN operands (quiet and signaling NaNs).

As it happens, Java defines its <CODE>min</CODE> and <CODE>max</CODE> methods for floating-point operands the same as the basic  **minimum** and **maximum** operations expected in the forthcoming IEEE 754.  One option would be to adjust the definitions of FMIN and FMAX to match that of Java.  If that were done, then the altered definitions would have FMIN and FMAX propagate NaNs exactly the same way that FADD and other floating-point instructions already do.  For example, instead of FMIN of *x* and NaN writing *x* to the destination register, it would set the destination to NaN.

jhauser-ucberkeley, 2017-05-19:
To be a little clearer perhaps:

The current version of IEEE 754 defines operations **minNum** and **maxNum**, and RISC-V's FMIN and FMAX are compatible with those existing definitions.  It appears that the next revision of IEEE 754 will jettison **minNum** and **maxNum** and replace them with a different set of **minimum** and **maximum** operations (including **minimumNumber** and **maximumNumber**, among others).  None of the new IEEE operations will be identical to the old **minNum** and **maxNum**, so none will be the same as FMIN/FMAX either, unless the RISC-V instructions are changed.

sorear, 2017-05-20:
Do we know yet what ISO C and C++ plan to do regarding the `fmin` and `fmax` functions, which are specified to match the present IEEE *minNum* / *maxNum* ?

jhauser-ucberkeley, 2017-05-20:
I don't know about C++, but the way the C Standard is defined, it will already be compatible with the IEEE's new **minimumNumber** and **maximumNumber** operations without any change to the language standard.  That's because the only difference between the old **minNum**/**maxNum** and new **minimumNumber**/**maximumNumber** concerns signaling NaNs, whereas C officially leaves the behavior of signaling NaNs unspecified.  Therefore I assume C will simply leave their specification for <CODE>fmin</CODE>/<CODE>fmax</CODE> unchanged and claim it now maps to **minimumNumber**/**maximumNumber**.

sorear, 2017-05-20:
C++ copies fmin and fmax from C by reference

The C99 draft I'm looking at says that `fmin(qNaN, 5) = 5`: §7.12.12.3 footnote 213 in N1256, "NaN arguments are treated as missing data: if one argument is a NaN and the other numeric, then the
fmax functions choose the numeric value. See F.9.9.2."

ECMA-262 requires `Math.min(qNaN, 5) = qNaN`: https://tc39.github.io/ecma262/#sec-math.min "If any value is NaN, the result is NaN."

Consider this less a specific concern and more a vague question of how IEEE 754-201X & RISC-V handle the conflicting requirements of C / C++ and Java / JS.

aswaterman, 2017-06-03:
Closing this, as it is resolved in https://github.com/riscv/riscv-isa-manual/pull/75

closed 2017-06-03 by aswaterman