riscv/riscv-isa-manual #69: Clarify upper 32 bits of result of FCVT.WU.S/D/Q in RV64

jhauser-ucberkeley, 2017-05-20:
In RV64, do instructions FCVT.WU.S/D/Q zero the upper 32 bits of the destination, or propagate bit 31?

aswaterman, 2017-05-20:
The latter.  Addressed in 03f0e54ee7203174b2778b91e9d5089ad9959ad9.  Thanks!

closed 2017-05-20 by aswaterman

aswaterman, 2025-08-19:
Yes, see the NOTE at the top of this section: https://github.com/riscv/riscv-isa-manual/blob/3be9f495acc280183ddc8dae6ed3dee72b71c56a/src/rv64.adoc#integer-computational-instructions

It is consistent with the handling of unsigned 32-bit integers in the base ISA and calling convention.

LWU is normally not used for handling 32-bit values, unless the intended operation is to take an unsigned 32-bit type and use it in 64-bit computations (e.g. `my_u64 + *my_u32_pointer` would use LWU, but almost any other use of `*my_u32_pointer` would use LW).