riscv/riscv-isa-manual #78: Ordering of IO Accesses and Timer Reads

palmer-dabbelt, 2017-06-03:
As part of the Linux upstreaming process, it was pointed out that some devices have timing constraints on register writes.  The specific example was

````
  writel(1, reset_reg);
  readl(reset_reg); /* flush posted writes */
  udelay(10);
  writel(0, reset_reg);
````

The read isn't strictly necessary, it appears just to be the API Linux provides for IO write ordering (the fences are before the write and after the read).  On RISC-V, this compiles to something like

````
  fence io,o
  st 1, RESET_REG
  ld RESET_REG
  fence o,io
loop:
  rdtime
  blt loop
  fence io,o
  st 0, RESET_REG
````

Since the fence instruction is only defined as ordering between the IO accesses, it doesn't actually prevent the processor from ordering the store to execute before the delay.  It might be sufficient to claim that this is disallowed because speculative writes can't be made visible (though is the write really speculative if you know it'll issue eventually?), but another option would be to include the timer as part of the IO space.

@aswaterman had mentioned that you'd been kicking around the idea and that I should open a bug.

aswaterman, 2017-06-03:
@kasanovic we should write down which CSR accesses are ordered by the FENCE I/O bits.  It's pretty clear that reading time/stime should be treated as device input.

kasanovic, 2018-11-07:
I think reading time should be treated as a serializing operation - i.e., you don't speculate around it.  This would not then need the FENCE.  If we didn't make it serializing, I can't see many cases where you could sensibly use the RDTIME without FENCEs.

aswaterman, 2018-11-07:
Yes.

kasanovic, 2018-11-07:
We do not have a framework for describing serializing instructions, in particular, their impact on the memory model.  If RDTIME is serializing, and the spec constraint of within 1 ULP across harts is preserved, then timestamp values will effectively impose extra ordering constraints on the memory system.

aswaterman, 2018-11-30:
Conclusion is that CSRs other than the hart itself should be ordered as I and O for purposes of FENCE.

(Note this really only applies to CSRs that can be written by actors other than the local hart, because otherwise you'd be none the wiser.)

aswaterman, 2019-03-13:
Closed by d07bea568fb33c26e81b19d2965fc488ab4c0602

closed 2019-03-13 by aswaterman