riscv/riscv-isa-manual #79: Reserve sstatus/mstatus bits for nonstandard extensions

aswaterman, 2017-06-03:
mstatus in RV32 is very cramped, but we want to reserve a few bits for nonstandard extensions.  I propose reserving just two (bits 30:29), keeping bits 28:23 reserved for standard extensions.

In RV64, we should reserve a more generous amount for nonstandard extensions; I propose the 16 bits 31:29 and 62:50, keeping bits 49:36 and 28:23 reserved for standard extensions.

(Originally suggested by Allen Baum)

aswaterman, 2018-11-21:
The immediacy of this concern is salved by a forthcoming proposal to add mstatush to RV32.  mstatush will contain much the same thing as mstatus[63:32] would on RV64.

The bits will remain reserved and can be allocated to nonstandard extensions later.  (Assuming that is, in fact, the right thing to do.)

closed 2018-11-21 by aswaterman

asb, 2018-11-28:
What do you think about reopening this issue and renaming to reflect the intent to add mstatush or creating a new issue to track that intended proposal? It's helpful for the rest of us if these things are easily discoverable via the issue tracker.