riscv/riscv-isa-manual #81: Startin/stopping all performance counters at one instruction

kasanovic, 2017-06-04:
soberl@nvidia.com
"I have another question for the perf. counters. Maybe I miss something, but I don't find a single CSR to start/stop all counting at same instruction.
If I have to stop counting by setting event selector to 0, different counters get different start points.
For m{h/s/u}counteren, looks it's for privilege controlling, thus not defined for machine mode.

I think either shared EN bit for all counters, or separate bits in shared CSR is OK for me."

ccelio, 2017-06-04:
Aren't event selectors only valid in machine mode anyways, so that technique may not work?

Is snapshotting at different start points an enormous deal since it's ~10 instructions from the start of the snapshot block to the end (the potential serialization to read/enable/disable the counters could be a larger effect).  Unless an interrupt comes in while snapshotting the counters, at which point atomically enabling/disabling the counters to measure fine-grain performance seems to be the least of the problems.

chuanhua, 2018-11-08:
For **power saving** purpose, sometimes we want all counters to be stopped, or sometimes we want just 1~2 counters to count. So we want to have a CSR with separate counter bits to disable or enable counting of counters.

chuanhua, 2018-11-08:
According to the privileged spec, writing 0 to the counter event selection CSR means "no event" and can be used to disable the counter. However, mcycle and minstret do not have the corresponding event selection CSRs. And we want these two counters to be disabled as well. Thanks.

aswaterman, 2018-11-21:
We are proposing following @chuanhua's suggestion to provide an mcountinhibit CSR to stop the various counters from incrementing.

As @ccelio points out, this is of no direct help to less-privileged modes, but that is an issue for another day.

aswaterman, 2018-11-21:
https://github.com/riscv/riscv-isa-manual/commit/9ffebdbb329d43893d62df5af47ae1a4602a18cc

closed 2018-11-21 by aswaterman