riscv/riscv-isa-manual #83: move pre-built PDFs out of the git repo to github releases

pabs3, 2017-06-12:
Please move the prebuilt PDFs out of the git tree:

https://github.com/riscv/riscv-isa-manual/tree/master/release

And put them in the github releases:

https://github.com/riscv/riscv-isa-manual/releases

VCS repositories should contain source code only, github created the releases area for prebuilt files.

aswaterman, 2017-06-12:
Agreed.  @palmer-dabbelt do you mind creating releases for all the PDFs in /release?

palmer-dabbelt, 2017-06-13:
Usually how we'd handle this is to:

 * Create a tag for the old release.
 * Attach the binaries to that tag on Git Hub.

but we don't have the commits for the old releases, so we don't have any tags.  I can just make tags that point to an empty repo, but that's a bit awkward.

pabs3, 2017-06-14:
If you still have copies of the source files for the old releases, then you could create new parentless commits for those and tag them.

    git checkout --orphan ancient_history
    git rm -rf .
    <unpack old source tarball version2>
    git add .
    git commit --date=<release date> -m 'Import <tarballname>'
    git tag <version>
    rm -rf *
    <unpack old source tarball version2>
    git add .
    git commit --date=<release date> -m 'Import <version2>'
    git tag <version2>
    git branch -D ancient_history

You can also graft those onto the history too:

https://git.wiki.kernel.org/index.php/GraftPoint
http://ben.straubnet.net/post/939181602/git-grafting-repositories

If you don't have the old source files then I think just creating tags pointing at empty orphaned commits seems like a reasonable way to do it. You can annotate the tags to make it clear what they are for.

pabs3, 2018-11-07:
Was this issue closed accidentally or are you rejecting the idea of moving the pre-built specification PDFs out of the git repo into github releases?

aswaterman, 2018-11-07:
Reopening.  I think it was accidentally closed because it was conflated with a related issue (automatically building the PDFs when the repo is pushed).

aswaterman, 2018-11-07:
(but actually taking action is up to @palmer-dabbelt :))

palmer-dabbelt, 2018-11-07:
I don't think we even have the source for those old releases, unless Andrew has some on an old machine somewhere?

aswaterman, 2018-11-07:
@palmer-dabbelt you should still have access to https://github.com/ucb-bar/riscv-doc right?

asb, 2018-11-30:
The releases/ directory is actually the easiest way of finding old releases though (and even then it's incomplete, missing v2.0 and v1.0 which you have to google for). Having automatically built draft versions of the spec is helpful, but it does mean that there are many many releases + tags making it pretty difficult to search through and find numbered releases on the github interface.

I don't know where to log it, but there's probably room for improvement on https://riscv.org/specifications/ too - it could link to the releases/ directory (or somewhere else?) to make it easier to find older versions of the specs.

kiniry, 2018-11-30:
I am particularly keen on cleaning up this and related public messes (IMO), such as the continual questions about the various `gnu` git repositories under https://github.com/riscv.  We look forward to working with you, @asb, and active contributors to those repositories of their upstream daughters on such.

aswaterman, 2018-12-21:
I've added v2.0 and v1.0 versions of the specs to the release directory.

I think the right solution is to move them all to https://riscv.org/specifications/, rather than putting them in github releases.

pabs3, 2018-12-21:
Moving them to the website would be fine too, the main thing is to not
have pre-built files in the source branches of the git repository.

-- 
bye,
pabs

https://bonedaddy.net/pabs3/

aswaterman, 2018-12-21:
Understood.

pabs3, 2018-12-21:
BTW, one advantage of having the specs on github releases would be to
have a place to store old versions of the specs. I think the specs
should definitely be on (or linked from) the website too though.

-- 
bye,
pabs

https://bonedaddy.net/pabs3/

aswaterman, 2018-12-21:
I think the RISC-V web site is also an acceptable place to store old versions of the specs.  To avoid confusion, they can be under a separate section than the current versions.