riscv/riscv-isa-manual #86: build figures from source, use non-proprietary source format

pabs3, 2017-06-12:
The `src/figs/*.pdf` files were created from `src/graffles/*.graffle` but the build system in `build/Makefile` does not rebuild the PDF figures when the OmniGraffle files change.

I would recommend to:

* remove the prebuilt PDF figures from the git repository
* switch from the proprietary OmniGraffle file format as source for your figures to an open format like SVG that can be converted to PDF at build time.

It looks like OmniGraffle can export to SVG, which can then be edited in vector graphics editors like Inkscape and then Inkscape can convert to PDF for import into the TeX document:

https://en.wikipedia.org/wiki/OmniGraffle
https://tex.stackexchange.com/questions/2099/how-to-include-svg-diagrams-in-latex

aswaterman, 2017-06-12:
We included the PDFs and don't automatically regenerate them because we don't expect people to have OmniGraffle installed.

There's loss of metadata when exporting from .graffle to .svg, so we'd still need two sets of files committed.

pabs3, 2017-06-13:
By sticking to OmniGraffle, you also prevent people not using it from improving/adjusting the diagrams or adding new ones.

I'm not very familiar with TeX diagrams, but is there any other non-proprietary source format that would preserve the mentioned metadata so that you could use it instead of OmniGraffle?

I'd like to make RISC-V a first-class architecture in Debian, including having the architecture specifications available to read offline. Unfortunately due to the use of the proprietary OmniGraffle format instead of a more open format, this means the riscv-isa-manual package would have to be put into Debian contrib instead of Debian main.

aswaterman, 2017-06-13:
Apparently @kasanovic has authored the only three graffles, so he should make the call.

Are the PDF figures themselves a problem for debian main, or is that a matter of your preference?

pabs3, 2017-06-13:
I'm not sure if it is written down anywhere but the Debian ftp-master's policy is that everything in main must be buildable from the source format using only other packages from main. The package build process is allowed to skip building them (if they take days to build for example) but they must be buildable.

aswaterman, 2017-06-13:
I guess the question is, is PDF an acceptable source format for images?  I have to assume that some binary formats are OK, e.g. bitmaps, at least for legacy reasons.

pabs3, 2017-06-13:
The source is the form that you as the upstream developers prefer for creating/modifying the figures. In this case it is clearly the OmniGraffle format that is your currently preferred format and the PDF form is clearly not.

aswaterman, 2017-06-13:
Yeah, that makes sense.

pabs3, 2018-01-16:
@kasanovic: any thoughts on alternatives to OmniGraffle for the source format for the riscv-isa-manual figures?

PR #156 (2018-04-13): Inkscape figures
Inkscape seems more than capable enough to maintain these images going forward; Fixes #86.
closed 2018-04-13 by aswaterman