riscv/riscv-isa-manual #93: Missing acronym definitions

ilg-ul, 2017-07-17:
I'm a novice to RISC-V and just started to read the specs.

The documents contain lots of acronyms, which is not a problem as long as the acronyms are known, but impedes the understanding if they are less known; for example Volume I mentions "RTL", (RTL designs, RTL implementation, etc), without ever defining it.

Could you either define RTL at the first reference, or even add a table of acronyms?

Thank you,

Liviu

ilg-ul, 2017-07-17:
Acronyms I had to google for:

- [RTL](https://en.wikipedia.org/wiki/Register-transfer_level)
- [ABA problem](https://en.wikipedia.org/wiki/ABA_problem)

DSHorner, 2017-07-19:
I agree, both could use explanation.
I will draft a revision.

ilg-ul, 2017-07-20:
yeah, it's not big deal, but since I noticed them I thought it might be worth reporting. though there might be other too...

sorear, 2017-07-20:
The ABA sentence should probably just be removed.  RISC-V implements "restricted LL/SC", which is exactly equal in power to CAS and does not have any ABA advantages.

aswaterman, 2017-07-20:
This is the paper that @sorear referenced last year: http://citeseerx.ist.psu.edu/viewdoc/summary?doi=10.1.1.140.8385

DSHorner, 2017-07-20:
What I appreciate about the manual is the justifications given in the comments. The ABA discussion is such an explanation of why lr/sc was chosen and CAS rejected. I think the section is valuable to retain. And if it goes, what else should also go?

ilg-ul, 2017-07-20:
> What I appreciate about the manual is the justifications given in the comments.

yes, me too. the "why?" story is a valuable source of info, not only for historians, but for us, mere mortals, too. and comparisons with other solutions are also useful.

my only frustration is sometimes the very specialised language, which makes the text somehow hard to read.