riscv/riscv-isa-manual #94: Consider changing C.MV expansion from add to addi or add commentary

michaeljclark, 2017-07-20:
The spec currently states:

> "C.MV copies the value in register rs2 into register rd. C.MV expands into add rd, x0, rs2."

Discussed with @sorear regarding the possibility of having C.MV expand to the canonical move. I recently tried this in a software implementation (spike and riscvemu interpret the compressed instructions whereas rv8 expands them into base ISA micro-ops perhaps similarly to how they may be handled in a micro-architecture with a reduced set of internal micro-ops; it makes macro-op fusion pattern matching easier as we can match expanded RVC with base ISA patterns). Tested in software:

https://github.com/rv8-io/rv8/commit/fc0d8c5b1b1c6ce5e7c9582638f4d6afd3cb5e6c

The possibility is to change the expansion to the canonical move:

> "C.MV copies the value in register rs2 into register rd. _C.MV expands into addi rd, rs2, 0_"

The only _slightly odd thing_ with the alternate expansion is a lane crossing from rs2 to rs1. C.ADD, C.FSDSP and C.FSWSP are the other RVC instruction that decode rs2 from bits[6:2]. In-order machines without a renamer might just issue an add and BOOM doesn’t implement RVC yet. The impact of the change is to add wire crossings in an OoO with RVC but then the renamer needs just one move pattern (it also has to wire cross to read the move operand from rs2 instead of rs1). Thus it is a benign change that doesn't affect backwards-compatibility, however, someone implementing based on the spec would end up with a single canonical move if the text is changed. Implementors may otherwise choose to expand it to addi rd, rs2, 0 (not sure why that is beneficial though). The alternative is perhaps commentary and/or text with two expansion options.

> "C.MV copies the value in register rs2 into register rd. C.MV naturally expands into add rd, x0, rs2 or alternatively; swapping rs1 and rs2; it expands to addi rd, rs2, 0; the mv pseudo-op"

The benefit of this third option is that it makes an implementor aware that two main patterns need to be detected for the renamer (there are in fact many more endogenous moves in the ISA, but they are not emitted by the compilers).

michaeljclark, 2017-07-20:
Note this option only requires a _slight_ text change (or commentary). It seems that the MV pseduo-op could have been `add rd, x0, rs2` so that RVC and the Base ISA use the same register slots for moves but that alternative is a somewhat more intrusive as binaries and MV use ADDI rd, rs1, 0. The expansion seems to be the right place to unravel RVC into a micro-op, and pointing this out in the text makes the two MV encodings clear to implementors.

aswaterman, 2017-07-20:
I view this as a purely microarchitectural issue.  There's no requirement
that C.MV be implemented by expansion to ADD; implementations are already
free to expand it to ADDI instead, or even avoid explicit expansion to
32-bit instructions altogether.

michaeljclark, 2017-07-20:
I agree, ... but it does not preclude a text change with suggested expansion options. I followed the text and ended up with two move patterns. If the text says it expands to ADD but implementations expand it to ADDI I would argue the spec could do with some minor commentary. My implementation now deviates from the wording in the spec, so the 3rd wording option is a loosening and essentially commentary (thus not an issue with the base ISA being frozen).

michaeljclark, 2017-07-20:
Something along these lines:

> "C.MV copies the value in register rs2 into register rd. C.MV naturally expands into add rd, x0, rs2 or alternatively; swapping rs1 and rs2; it expands to addi rd, rs2, 0; the mv pseudo-op"

michaeljclark, 2017-07-20:
or the same text but with some blockquote on the page about MV pseudo-op vs C.MV having different expansions. If expanding to ADDI is an expansion option that is in contrast to the stated expansion, then I think it deserves a sentence in the spec. If someone follows the spec expansions they will end up with two different moves. Alternatively, we could list two mv pseudos which is essentially what we have if we follow the spec expansions:

mv  addi rd, rs1, 0
mv  add rd, zero, rs2

It's a minor usability issue that I think deserves a minor comment in the spec.

michaeljclark, 2017-07-20:
How about this?

"C.MV copies the value in register rs2 into register rd. C.MV naturally expands into add rd, x0, rs2 or alternatively; if rs1 and rs2 are swapped; C.MV could expand to addi rd, rs2, 0; the mv pseudo-op (*)"

> It is a micro-architectural concern as to how an implementation expands C.MV however it is worth noting that the Base ISA mv pseudo-op expands to addi rd, rs1, 0 while C.MV naturally expands to add rd, zero, rs2 due to it sharing the same encoding as C.ADD. This detail is relevant for micro-architectures that implement moves as register renames. The ISA contains many endogenous move instructions, however, C.MV, ADDI rd, rs1, 0 and ADD rd, x0, rs2 represent the set of canonical move operations that are expected to be emitted by compilers; the last form based on the C.MV expansion noted here in the main text.

michaeljclark, 2017-07-20:
Neither Rocket nor BOOM has this issue as one doesn't rename and the other doesn't RVC. We should tag as "usability" i.e. so one can discover it from the text rather than cosmetic bug reports :-D

aswaterman, 2017-07-20:
I'm OK with adding a short commentary note to this effect:

C.MV expands to a different instruction than the canonical MV pseudoinstruction, which instead uses ADDI.  Implementations that handle MV specially, e.g. using the register-renaming hardware, may find it more convenient to expand C.MV to ADDI instead of ADD, at slight additional hardware cost.

michaeljclark, 2017-07-20:
I like that. I think it addresses a usability issue.

michaeljclark, 2017-07-20:
it is mainly the reader discovering that there are three forms of MV for the renamer, the mv pseudo op (addi), C.MV and the expansion in the text for C.MV (add). Yes, it could be handled in expand or as two or even three patterns for rename detection depending on how the implementor decides to do it. The renamer has to have wires to either rs1 or rs2 so I don't think there is a cost difference. i.e. it's semantics. It would be less wires if mv pseudo was add rd, zero, rs2; but one could argue that the renamer should detect all of the endogenous moves (although useless wires not touched by the compiler).

The usability issue is that I was totally unaware that C.MV did not expand to the same instruction as MV pseudo. I assumed. That's all that really needs to be addressed.

aswaterman, 2017-07-20:
I agree that simply noting they are different is most important.  I think some readers may like to know why it matters why they are different.

aswaterman, 2017-07-20:
OK added

closed 2018-11-06 by kasanovic