riscv/riscv-isa-manual #101: vdcfg explanation

respasa, 2017-08-31:
I would find it easier to read & understand the document if it said the following:

There is a 256b configuration CSR for the vector unit called vdcfg that holds all the vemaxw and vetype information, with bits $8n$ to $8n+4$ holding {\tt vetype}$n$ and bits $8n+5$ to $8n+7$ holding {\tt vemaxw}$n$. Alias CSRs are provided to read and write this information in RV32, RV64 and RV128 as described below. 

I.e., state clealry that vdcfg is a 256b register and that there are alias CSRs to read it in the various modes.... The current text requires a bit more intellectual effort :-) to read

kasanovic, 2017-09-15:
Agree.

kasanovic, 2018-11-07:
outdated now, closing.  All future comments should be on V spec repo.

closed 2018-11-07 by kasanovic