riscv/riscv-isa-manual #103: FNMADD and FNMSUB mistake?

mn416, 2017-09-05:
Manual says:

> FNMSUB.S computes -rs1 * rs2 + rs3; and FNMADD.S computes -rs1 * rs2 - rs3.

Other way round?

aswaterman, 2017-09-05:
The manual is correct. The confusing names come from MIPS, where FNMSUB was defined to compute -(rs1*rs2-rs3). That is, they defined the negation to happen after computing the sum.

mn416, 2017-09-05:
Ah, thanks!  My code was passing the test suite, but it turns out that the RISC-V instruction encoding actually made me implement the correct behaviour without realising it :)

closed 2017-09-05 by mn416

tovine, 2018-03-08:
@aswaterman might I suggest adding parentheses in the text to clarify the grouping of the numbers? I too implemented it wrongfully in the first place after reading too quickly; it would have been no room for misinterpretation if it said `[+/-](rs1 x rs2) [+/-] rs3` or something similar, even if it is technically correct as it is now.

tovine, 2018-03-08:
Additionally, the [riscv-tests](https://github.com/riscv/riscv-tests/blob/master/isa/rv64uf/fmadd.S) for these instructions get it wrong then; expecting -3.5 with inputs 1.0, 2.5, 1.0 for `FNMADD`

I can fix the tests and submit a pull request if it is confirmed that the `N` means negating the product before computing the sum.

aswaterman, 2018-03-08:
Let me double-check that last claim, and think about ways we might clarify this. We could potentially add some commentary to explain the seeming inconsistency.

tovine, 2018-03-08:
I re-read it a bit too quickly and misinterpreted the manual, it's been a long day I guess...
After starting with the intuitive interpretation of the name (FNMADD - multiply and add, then negate the result) and doing the algebra I can see that it matches the description in the manual.

Making a little note about it  is probably a good idea though.

tovine, 2018-06-06:
>Fused multiply-add instructions multiply the values in rs1 and rs2, optionally negate the product, then add or subtract the value in rs3, writing the final result to rd. FMADD.S computes rs1 * rs2+rs3; FMSUB.S computes rs1 * rs2-rs3; FNMSUB.S computes -rs1 * rs2+rs3; and FNMADD.S computes -rs1 * rs2-rs3.

I think the text in this section is somewhat misleading:
>multiply the values in rs1 and rs2, **optionally negate the _product_ (=a * b), _then_ add or subtract** the value in rs3

When interpreting that sentence, it clearly (at least in my opinion) yields the following for the negated versions:
>FNMADD = `-(a * b) + c`
FNMSUB = `-(a * b) - c`

Instead of the actual behavior:
>FNMADD = `-((a * b) + c) = -(a * b) - c`
FNMSUB = `-((a * b) - c) = -(a * b) + c`

Hence I think the wording should be changed to something like this:
>Fused multiply-add instructions multiply the values in rs1 and rs2, _**then add or subtract the value in rs3 and optionally negate the result before writing it to rd**_. FMADD.S computes rs1 * rs2+rs3; FMSUB.S computes rs1 * rs2-rs3; FNMSUB.S computes -rs1 * rs2+rs3; and FNMADD.S computes -rs1 * rs2-rs3.

@aswaterman what do you think?

tovine, 2018-06-06:
But is there not any way to write it that would be true and a bit less confusing?
Isn't the following statement different from the actual behavior?
>multiply the values in rs1 and rs2, optionally negate the product, then add or subtract the value in rs3

How is the behavior described there the same as this?
>FMADD.S computes rs1 * rs2+rs3; FMSUB.S computes rs1 * rs2-rs3; FNMSUB.S computes -rs1 * rs2+rs3; and FNMADD.S computes -rs1 * rs2-rs3.

I'm sorry that I keep nagging on this, but I get just as confused every time I read that section of the manual, I can't be the only one? 😕

aswaterman, 2018-06-06:
I really do not see the inconsistency between the two statements.  FNMADD definitely fits the description of "optionally negate the product, then add or subtract the value in rs3."  -- it just doesn't make the choice of "add or subtract" that you'd expect from the instruction's name.

aswaterman, 2018-06-06:
But to be clear, I'm still open to rewriting this paragraph in a less confusing way, as long as it doesn't introduce this inaccuracy about negating the sum.

tovine, 2018-06-06:
Hmm, I see now (again) that you are correct.

My confusion was probably from attempting to think about the process in the order that I read it and it just intuitively felt strange.
Maybe putting the `FMADD.S computes...` information in a table or something similar to make it immediately stand out from the text would help? That worked really well for the table about the domains of float-to-integer conversions and behavior for invalid inputs (Table 8.4 in version 2.2 of the user ISA spec).

Another (somewhat related) question: would the result still be negative zero for `-0*0+0`, or would the add operation discard the negative sign of the product? (I'm asking because this is a corner that the PULP FPU may have skipped)

aswaterman, 2018-06-06:
Not opposed to the table idea, but I don’t have time to make it today.

The case you mentioned should give +0.  To get a -0 result, make both the product and addend negative (e.g. initialize ft0 to +0 then do FNMADD with ft0 for all three operands).

aswaterman, 2018-06-18:
Yeah.

asb, 2018-06-18:
I quite like the verbose and explicit description in @tymcauley's comment - I suggest just inserting that into the spec. Concision can sometimes aid readability, but given the confusing naming of these instructions (that we can blame on MIPS!) I don't think that's the case here.

tymcauley, 2018-06-18:
Agreed. While we're editing these descriptions, one other clarifying idea to bring up: adding parentheses around the products in these four equations. It is clear from standard order-of-operations (multiply before add/subtract) that _a_ * _b_ + _c_ means that one should compute _a_ * _b_, then add that product to _c_. However, I would argue that (_a_ * _b_) + _c_ is still easier to read. All ambiguity is removed, and you don't even have to think about order-of-operations. This probably seems like a silly distinction to a lot of hardware/software engineers that are used to dealing with IEEE floating point and are familiar with FMA operations, but I imagine a reasonable number of people that aren't familiar with these operations might go through this spec, and this could help remove some confusion for them.

So, if we did make that change, the sentences might look like this:

> FMADD.S multiplies the values in _rs1_ and _rs2_, adds the value in _rs3_, and writes the final result to _rd_. FMADD.S computes (_rs1_ * _rs2_) + _rs3_. FMSUB.S multiplies the values in _rs1_ and _rs2_, subtracts the value in _rs3_, and writes the final result to _rd_. FMSUB.S computes (_rs1_ * _rs2_) - _rs3_. FNMSUB.S multiplies the values in _rs1_ and _rs2_, negates the product, adds the value in _rs3_, and writes the final result to _rd_. FNMSUB.S computes -(_rs1_ * _rs2_) + _rs3_. FNMADD.S multiplies the values in _rs1_ and _rs2_, negates the product, subtracts the value in _rs3_, and writes the final result to _rd_. FNMADD.S computes -(_rs1_ * _rs2_) - _rs3_.

Once we figure out how we want to proceed, I'm happy to put together a PR to take care of it.

aswaterman, 2018-06-18:
Sounds good.  Can you make a PR?