riscv/riscv-isa-manual #106: Possible clarification for what's written to mepc for ecall/ebreak?

maikmerten, 2017-09-20:
From the privileged spec 1.10, section 3.1.19 (page 34):

"When a trap is taken into M-mode, mepc is written with the virtual address of the instruction that
encountered the exception."

After reading this I was still unclear about what the content of mepc should be in the case of ecall/ebreak.
 - if ecall/ebreak is the instruction that "encounters" the exception, then mepc should contain the address of that ecall/ebreak instruction. In this case mepc will *not* denote where execution should resume: instead mepc needs to be loaded, incremented by 4 and updated accordingly before a mret can be executed.
 - if ecall/ebreak are merely "setting up" the exception, then the succeeding instruction may be the one that "encounters" them. In this case mret can be called directly.

According to IRC conversations, it appears that the first case is the intended one. Perhaps it's helpful to some users to clarify this in the manuals.

For other types of exceptions (e.g. external interrupts or timers) it appears that mepc will contain the address where execution should resume after exception handling (i.e., a "proper" return address for mret to use).

maikmerten, 2017-09-21:
Thanks for pointing this out, this indeed clarifies things.

FrankBuss, 2018-11-18:
@aswaterman That's nice, but it is still missing in the "User-Level ISA Specification" for the ECALL description. Also there is some redundancy, because ECALL is described in the privileged specification as well, but a little bit different.
It would be nice to have one place, which describes all aspects of the program flow: How the PC is normally incremented, how ECALL works, how exceptions work, and how interrupts work, and what CSRs are set and read, and how the PC is changed. At the moment it is very difficult to gather this information from all the different places in both specifications.

aswaterman, 2018-11-19:
I don’t agree that this detail belongs in the unprivileged spec.

I do agree that a single localized description of what happens on an ECALL or other exception would be an improvement, as it is currently decentralized.

closed 2018-11-19 by aswaterman