riscv/riscv-isa-manual #108: Specify behavior of satp w.r.t writable-XLEN

aswaterman, 2017-09-22:
When running an RV32 S-mode on an RV64 M-mode, how does M-mode perceive the value of satp?

Current proposal is to project RV32 satp onto RV64 satp as follows:
RV32 satp.ppn -> RV64 satp.ppn, zero-extended
RV32 satp.asid -> RV64 satp.asid, zero-extended
RV32 satp.mode[0] -> RV64 satp.mode[3], with bits 2:0 zero-filled

And the converse question: if RV64 M-mode sets a high bit of satp.ppn/satp.asid or a low bit of satp.mode, is it simply ignored in RV32 S-mode?

Also make sure all the other CSRs' projections are trivial, and define the projection if not.

sorear, 2017-09-22:
The problem also exists for m-registers, since M-mode can change its own XLEN, and also the question of what happens when you try to xret with a noncanonical xepc.

Are we specifically trying to avoid declaring anything unpredictable?

aswaterman, 2017-09-22:
It would be nice to keep all of the behavior well defined, but I don't view that as essential.

acw1251, 2017-09-25:
Andrew, I like the projection, but would it be possible to to show this in the manual as a third view of satp for M-XLEN = 64 and S-XLEN=32 (or in other words its an RV64 csr view of an RV32 satp)? It would look something like this:

[ MODE (WARL) | 00000000 | ASID (WARL) | 000... 22 zeros ...000  | PPN ]

By showing it as a different view instead of projecting it into the RV64 satp, you don't have to worry about the WARL property of the RV64 satp.mode field (avoiding problems where you only implement Sv32 and Sv48).

Alternatively the width of the satp CSR can be defined as S-XLEN instead of XLEN, so it is a 32-bit CSR whenever s-mode is 32-bit, and if it is being accessed with XLEN=64, then it is sign extended (or whatever is done for accessing mcounteren when XLEN=64).

aswaterman, 2017-10-06:
Defining satp to be S-XLEN bits wide makes sense (likewise other supervisor CSRs).  Zero-extension is the usual convention for CSRs.

@jhauser-ucberkeley has suggested we reconsider the obvious: simply projecting RV32-satp into RV64-satp[31:0].  His argument is that processors that have both writable-XLEN and virtual memory are likely to be fairly big, and will have 44-bit physical addresses anyway, or close enough that the extra couple flops don't matter.  I think it's a big enough win for simplicity that I agreed with him on the mailing list.

aswaterman, 2018-11-21:
This was resolved in cdb665618383ec40aeac722d0c99a3e46b974a4f and the conclusion was that the obvious projection (zero-extension / truncation) is the right thing to do.

closed 2018-11-21 by aswaterman