riscv/riscv-isa-manual #110: Clarify that, for mtval, "virtual address" sometimes means "physical address"

jhauser-ucberkeley, 2017-10-28:
For `mtval`, the document says
> When a hardware breakpoint is triggered, or an instruction-fetch, load, or store address-misaligned,
access, or page-fault exception occurs, `mtval` is written with the faulting virtual address.

However, when executing in M mode, the addresses that trigger hardware breakpoints or cause misaligned exceptions or access exceptions are actually physical addresses.  (And arguably this is true also when executing in U mode, if S mode and its address translation aren't supported.)  So, when the document says "virtual address" here, it doesn't always actually mean what it says.  This same flaw appears in multiple locations where the values that get written to `mtval` are documented.

(I don't think I'm the first to make this observation, but I don't see an existing ticket for it.)

aswaterman, 2017-10-30:
Isn't this just a definitional issue?  The implication is that the addresses really are virtual addresses, where the mapping from virtual to physical is the identity function.

If you don't like that hack, can you propose an alternative?

jhauser-ucberkeley, 2017-11-01:
My point is not to claim that the current way of handling this is necessarily wrong, only that the text could be confusing to the uninitiated.

Ordinarily, the whole point of virtual addresses is their independence from physical addresses, thanks to address translation.  Of course, it is always possible for the virtual-to-physical mapping to be the identity function.  But it is fair to say that a standard RISC-V without S mode has no support for virtual memory or address translation, and from there it is not at all unreasonable for a reader to ask:  What does it mean that "`mtval` is written with the faulting virtual address" when my RISC-V has no virtual addresses because it has no address translation?  The fact that in this case you mean for the "virtual" address to be identically the physical address is known to you, but not clearly communicated in the document, I believe.

Here's how I would fix the issue.  In 3.1.21, "Machine Trap Value (`mtval`) Register", I'd remove the word *virtual* from this sentence:
> When a hardware breakpoint is triggered, or an instruction-fetch, load, or store address-misaligned, access, or page-fault exception occurs, `mtval` is written with the faulting ~~virtual~~ address.

Instead, I'd add this sentence somewhere close by:
> When address translation is in effect, an address written to `mtval` is the virtual address before any address translation is applied.

Likewise, I'd remove the two instances of *virtual* from this paragraph:
> For misaligned loads and stores that cause access or page-fault exceptions, `mtval` will contain the ~~virtual~~ address of the portion of the access that caused the fault.  For instruction-fetch access or page-fault exceptions on systems with variable-length instructions, `mtval` will contain the ~~virtual~~ address of the portion of the instruction that caused the fault while `mepc` will point to the beginning of the instruction.

I've since noticed that the same issue applies to register `mepc`.  In 3.1.19, "Machine Exception Program Counter (`mepc`)", I'd remove *virtual* from the sentence,
> When a trap is taken into M-mode, `mepc` is written with the ~~virtual~~ address of the instruction that encountered the exception.

and add this sentence:
> When address translation is in effect, the address written to `mepc` is the instruction's virtual address before any address translation is applied.

Finally, in section 3.1.13, "Machine Trap Delegation Registers (`medeleg` and `mideleg`)", I'd remove *virtual* from this sentence:
> When a trap is delegated to a less-privileged mode *x*, the *x*`cause` register is written with the trap cause; the *x*`epc` register is written with the ~~virtual~~ address of the instruction that took the trap; ....

For `stval` and `sepc`, the existing text can be left alone, as the word *virtual* is no problem there.

aswaterman, 2018-11-21:
Sorry for sitting on this for a year, but I still think it makes the text less clear.  I'm open to introducing an alternative term like "original address", kinda like we did for the hypervisor spec.  But the onus is on you! :)

closed 2018-11-21 by aswaterman

aswaterman, 2018-11-21:
Problem is, it’s not just effective adddesses; it’s also program counter values.

kasanovic, 2018-11-21:
Can we just define PC to also be an effective address?

aswaterman, 2018-11-22:
To me, that is more confusing than my proposal from last year: keep using the term "virtual address" to mean "PC or effective address", but clearly define that "virtual address" doesn't imply address translation is enabled: when it's disabled, a virtual address is a physical address.

cc @jhauser-us