riscv/riscv-isa-manual #112: Delegating S-ECALL to supervisor (scause = 8)

as-sc, 2017-11-01:
machine.tex line 1115 says:
> In particular, {\tt medeleg}[11] and {\tt sedeleg}[11:9] are all hardwired to zero.

which implies that S-ECALL can potentially be delegated to S-mode. There is only one scause value for ecall (8) which corresponds to U-ECALL in mcause, no value for S-ECALL. While ecall source can be determined from SPP in supervisor trap, this seems inconsistent with mcause encodings.

aswaterman, 2017-11-02:
Yes, medeleg[9] should also be hardwired to 0.  (All implementations I'm aware of do so anyway.)

medeleg[10] should not be, as it is the mechanism to delegate ECALLs from guest OSes to the hypervisor when the hypervisor extension is present.  (The hypervisor spec proposal hasn't been released yet, so this fact is far from obvious.)

aswaterman, 2017-11-02:
Along the same lines, sedeleg[8] should be hardwired to 0, as it doesn't make sense to let a user program handle its own ECALLs.  Again, all implementations I'm aware of do so anyway.

aswaterman, 2017-11-02:
I guess the explanation in the spec does not justify this change, as it is predicated by "Some exceptions cannot occur at less privileged modes" -- and, of course, S-ECALL can occur in S-mode and U-ECALL can occur in U-mode.  The justification for hardwiring medeleg[9]=0 and sedeleg[8] is different--ECALLs should usually increase the privilege mode (except in M-mode, where they cannot).

closed 2018-11-21 by aswaterman

as-sc, 2019-01-14:
Seems that it hasn't been actually fixed.. There is also value 9 in scause description which is not reserved

AoteJin, 2021-06-21:
Yes, the value 9 in scause as a legal value is very confusing. Do we probably plan to get rid of it?