riscv/riscv-isa-manual #120: Resolve ambiguities for CSRs due to configurable XLENs

jhauser-ucberkeley, 2017-12-24:
In the face of separately configurable XLENs for M, S, and U modes, the privileged architecture needs to be clearer about what happens with CSRs when the XLENs of the various modes are not all equal, or when XLENs are changed.  Currently, most CSRs are documented as simply having width "XLEN", which is far too ambiguous in this environment.

First, I observe that the base ISA document already says that reading a CSR that is narrower than the effective XLEN returns a zero-extended value.  The main thing that's missing is to specify the widths of the CSRs more precisely.  To that end:

Change the specified widths of these registers from "XLEN" to "M-XLEN":

    misa
    marchid
    mstatus
    mtvec
    medeleg
    mideleg
    mip
    mie
    mscratch
    mepc
    mcause
    mtval
    mvtval

I recommend changing the width of `mvendorid` to 32.  There can be no advantage to a wider width for `mvendorid`.  Similarly, change the specified widths of each of these registers to either 32 or "M-XLEN":

    mimpid
    mhartid

Also, consider stating explicitly that the widths of `mcounteren` and `scounteren` are 32 bits, as appears to already be the case.

Change the specified widths of these registers from "XLEN" to "S-XLEN":

    sstatus
    stvec
    sip
    sie
    sscratch
    sepc
    scause
    stval

Change the specified widths of these registers from "XLEN" to "HS-XLEN":

    hstatus
    hedeleg
    hideleg
    bsstatus
    bsip
    bsie
    bstvec
    bsscratch
    bsepc
    bscause
    bstval
    bsatp

Most of the registers listed above are illustrated in figures whose width is indicated as "XLEN".  All these figures must be changed accordingly.

That should take care of questions about reading/writing CSRs for different XLENs.  There are still issues involving changing XLENs.  

Add a new section to Chapter 2, "Control and Status Registers (CSRs)", with this information:

> If the width of a CSR is changed (for example, by changing M-XLEN or U-XLEN), the values of the *writable* fields and bits of the new-width CSR are, unless specified otherwise, determined from the previous-width CSR as though by this algorithm:
>
> 1. The value of the previous-width CSR is copied to a temporary register of the same width.
>
> 2. For the read-only bits of the previous-width CSR, the bits at the same positions in the temporary register are set to zeros.
>
> 3. The width of the temporary register is changed to the new width.  If the new width W is narrower than the previous width, the least-significant W bits of the temporary register are retained and the more-significant bits are discarded.  If the new width is wider than the previous width, the temporary register is zero-extended to the wider width.
>
> 4. Each writable field of the new-width CSR takes the value of the bits at the same positions in the temporary register.
>
> Changing the width of a CSR is not a read or write of the CSR and thus does not trigger any side effects.

Note that the rules above say "unless specified otherwise".  Register `misa` is an exception to those rules.  For maximal clarity, in section 3.1.1, "Machine ISA Register `misa`", add this paragraph (or devise your own):

> If the value read from `misa` is nonzero, field MXL of that value always denotes the current M-XLEN.  If a write to `misa` causes M-XLEN to change, the position of MXL moves to the most-significant two bits of `misa` at the new width.

Exceptions to the general rules are also needed for fields SXL and UXL in `mstatus`.  In section 3.1.8, "Base ISA Control in `mstatus` Register", add before the last paragraph:

> If M-XLEN is changed from 32 to a wider width, each of `mstatus` fields SXL and UXL, if not hardwired to a forced value, gets an unspecified value from among the set of supported values.

If you don't like "gets an unspecified value", you can use this instead:

> ... gets the value corresponding to the widest supported width not wider than the new M-XLEN.

(Also, while you're there, please consider this adjustment:

> In particular, the implementation may hardwire UXL so that U-XLEN = M-XLEN or U-XLEN = S-XLEN.

This is not a new freedom, because section 4.1.2, "Base ISA Control in `sstatus` Register", already says that U-XLEN may be hardwired to S-XLEN.)

The CSR swapping done by the hypervisor extension also needs clarification when XLENs are not the same size, but I'll put that in a ticket to follow this one, since it may be dependent on how these issues get resolved outside the hypervisor extension.

aswaterman, 2017-12-27:
Thanks, John; this all seems well reasoned (though I don't favor unspecified behavior for UXL/SXL).  I will make a pull request this week to incorporate these suggestions.

jhauser-ucberkeley, 2017-12-27:
You may have noticed, I edited the text above a few times to add registers that I somehow forgot the first time around.  Hopefully I've got most of them now.  Not included in the specifics above are the N extension registers, because these are currently documented in detail only in the base ISA specification, where "U-XLEN" isn't a defined term.  Also, I don't know enough yet about the debug registers, `tselect` and `tdata1`-`tdata3` to comment on those.

pmundkur, 2018-03-14:
I'm trying to understand the implications of the above conversion algorithm, say specifically for satp.mode.

It appears that going from a wider sxl to an sxl of 32 should be ok, since the satp.mode for rv32 is just a bit, and both possible values are well-defined.

It's not clear to me what happens in the case of going from 32-bit sxl (with satp32.mode of Sv32) to 64-bit sxl.  One reading of the above algorithm says that the satp64.mode bits are zero, and so the new satp64.mode is Bare.  Is this a correct reading?

Semi-related to the above question is what happens in 64-bit S-mode when satp.mode contains an invalid value (e.g. one of the reserved values in RV64).  Could this happen as a result of changing SXL?  What kind of exception is generated when S-mode finds an invalid satp.mode?

jhauser-ucberkeley, 2018-03-15:
> It's not clear to me what happens in the case of going from 32-bit sxl (with satp32.mode of Sv32) to 64-bit sxl. One reading of the above algorithm says that the satp64.mode bits are zero, and so the new satp64.mode is Bare. Is this a correct reading?

That's how it's supposed to work, yes.

jhauser-ucberkeley, 2018-03-15:
> Semi-related to the above question is what happens in 64-bit S-mode when satp.mode contains an invalid value (e.g. one of the reserved values in RV64).

The spec says somewhere that if you attempt to write a value to satp that would result in an unsupported MODE field, satp is left entirely unchanged.  So it's not possible to get an unsupported value into satp.MODE.

> Could this happen as a result of changing SXL?

Not with just RV32 and RV64, no.

> What kind of exception is generated when S-mode finds an invalid satp.mode?

The situation can't occur with just RV32 and RV64.

If/when RV128 is fully fleshed out, the spec will probably have to say something special about how changing SXLEN from 128 to 64 affects satp.MODE.

mwachs5, 2018-04-13:
@jhauser-ucberkeley I saw your comment related to the debug registers.

Are there rules about these different XLENs? Like is HSXLEN <= MXLEN at all times, and so on?

mwachs5, 2018-05-03:
Please note that the Debug TG has responded to this change by updating the debug and trigger registers to be MXLEN wide. This does mean that if SXLEN  > MXLEN, you would run into a problem that the debugger can't, say, write the trigger registers to configure a virtual address trigger that covers the entire S-address space. But it seems like you would run into all sorts of problems when SXLEN > MXLEN, so not sure how to address that anyway.

(https://github.com/riscv/riscv-debug-spec/commit/675bb14656f6605fcbe303de0cb16cff08301c6e)

aswaterman, 2018-05-03:
IMO we should acknowledge this limitation and move on.

closed 2018-07-03 by aswaterman