riscv/riscv-isa-manual #129: MTVAL setting on ecall instruction

LeeMooreImperas, 2018-01-24:
Can you please give more information about how the trap value registers (e.g. mtval) interact with ecall instructions? Specifically, should an ecall cause the trap value register to be set, or not? From reading the specification (Privileged v1.10 Page 35, 3.1.21), I would expect that the trap value should be set either to zero or some other value, but in Spike it seems it is not updated at all on an ecall

Thx
Lee

aswaterman, 2018-01-24:
It should be set to zero, as you read from the spec.

IIRC I fixed this Spike bug fairly recently.  It was a holdover from an older version of the privileged architecture.

AntonKrug, 2018-06-21:
Hi @LeeMooreImperas 

Are you actually getting the content of the CSR while using mtval name? I Still get assembler errors when I try to read mtval CSR into a regular register. In meantime, I'm just using the older name for it the mbadaddr.

eroom1966, 2018-06-21:
@AntonKrug 
Hi Anton
I don't recall getting assembler errors, I think the issue was the runtime behavior of spike ?
Thx
Lee

AntonKrug, 2018-06-21:
@eroom1966 

You can use the older name for that CSR, it's the mbadaddr

closed 2018-11-06 by kasanovic