riscv/riscv-isa-manual #131: RV32 and 34 bit addressing

LeeMooreImperas, 2018-02-05:
I've just noticed that RV32 supports 34-bit physical addresses (see page 45 of the Privileged spec 1.10). These extra two bits are only accessible when virtual memory is enabled. I can't however find any description of what memory is accessed when Sv32 address translation is off. There are at least two plausible strategies:

1: allow access to only the low 4Gb (i.e. the top 12Gb are not accessible);

2; allow access to the bottom 2Gb if the MSB of the address is clear and the top 2Gb if the MSB is set (i.e. sign-extend addresses).

Additionally, if an RV64 processor is connected to a 56-bit physical bus (the maximum), what is the physical memory map when running with virtual memory disabled?

Thx
Lee

aswaterman, 2018-02-05:
Physical addresses are zero-extended, so it's option 1... but it does not seem we ever explicitly stated this.

LeeMooreImperas, 2018-02-06:
@aswaterman 
Hi Andrew, thanks for the answer, can you also confirm the behavior for the second part of the question

> Additionally, if an RV64 processor is connected to a 56-bit physical bus (the maximum), what is the physical memory map when running with virtual memory disabled?

Thx
Lee

aswaterman, 2018-02-06:
Everything's in the lower 2^56 bytes (i.e. the upper 8 bits must be zero).

closed 2018-07-06 by aswaterman