riscv/riscv-isa-manual #134: Should the mstatus MPP field be WARL ?

gameboo, 2018-02-13:
The `MPP` field is defined as `WLRL`. Should it be `WARL` instead?                                                                                                                                                                    

I understand that `WLRL` means that an implementation is at liberty to raise an exception or not on writes of an illegal value, and even to return an arbitrary bit pattern on reads, including illegal values such as the one of the previous write attempt. In the rv32/64mi illegal.S test, there is a comment saying "Skip the rest of the test if S-mode is not present", and it looks to me like the code there is using the `MPP` field of the `mstatus` register as a `WARL` field to test that the supervisor mode is available (that is it expects to be able to read back something strictly different than PRIV_S if the supervisor mode isn't implemented).

aswaterman, 2018-02-13:
Yes, v2.3-draft has redefined the xPP fields to be ```WARL``` (see commit 32b74b7a02e492c48a910ba0d20bcc52155b0616).

gameboo, 2018-02-13:
Ah! great, thanks.

closed 2018-02-13 by aswaterman