riscv/riscv-isa-manual #135: WFI + N virtualization hole

sorear, 2018-02-15:
The spec allows (is unclear, may or may not require) WFI to be executable in user mode if the N extension is available.  However, there is no S-mode equivalent of TW, so classical virtualization is no longer able to intercept WFI.

aswaterman, 2018-02-15:
Good point.

For systems with S-mode, ECALL is arguably a better way to do this sort of thing, in which case the equivalent of TW should always be in effect for U-under-S.

For M+U systems, where U-mode WFI is likely to be much more useful, we could use the existing mstatus.TW setting to control U-mode access to this feature.

aswaterman, 2018-03-09:
Krste that concurs WFI should always trap in U-under-S; mstatus.TW applies for U-under-M.

closed 2018-11-21 by aswaterman

aswaterman, 2018-11-29:
See #286