riscv/riscv-isa-manual #138: Memory architecture questions

JamesKenneyImperas, 2018-02-19:
Hello,

I have some questions and observations about memory support on RISCV architectures: specifically, I'm seeing two odd behaviors because of the way some features have been defined, and I want to check that these things are understood and expected. I've also got a question about the intended implementation of translation control bits such as mstatus.SUM and mstatus.MXR.

**Lower-privilege mode access to higher-privilege memory**
The first observation is that it is possible by misusing the Privileged Architecture control state to grant a low privilege mode access to any higher privilege mode memory. For example, this short sequence enters User mode with full read/write access to Supervisor mode memory:

        // force value 1 into mstatus.MPP (Supervisor mode)
        // force value 1 into mstatus.MPRV
        csrr    t0, mstatus
        li      t1, (3<<11)
        or      t0, t0, t1
        xor     t0, t0, t1
        li      t1, (1<<11) | (1<<17)
        or      t0, t0, t1
        csrw    mstatus, t0

        // return to User mode using *sret* instruction
        la      t0, 1f
        csrw    sepc, t0
        sret
    1:

Code after the label 1 is running in User mode but performs loads and stores as if in Supervisor mode (I've confirmed this using Spike). In a similar way, it is possible to grant Machine mode access to either Supervisor or User mode. Unless there is some reason I can't think of why this ability is required, I would recommend changing the behavior so that:
1. using the wrong exception return instruction is illegal (e.g. sret in machine mode); and/or:
2. changing the defined behavior of mstatus.MPRV so that it never allows granting access permissions to higher-privilege memory; and/or:
3. redefining the way that this is done by introducing new instructions that perform load and store with access rights associated with a lower-privilege mode (like the LDRT/STRT instructions in ARM) and removing the MPRV bit altogether.

**MPP TOR regions ending at top of memory**
A second observation relates to MPP TOR-type regions: as the regions are currently defined, it does not seem to be possible to create such a region that ends at the last memory address (for RV32). The maximum value that can be placed in a PMP address register is 0xffffffff, corresponding to the 34-bit physical address 0x3fffffffc, meaning any accesses to physical addresses 0x3fffffffc-0x3ffffffff (via a SATP-translated virtual address) can never be in range. Is this intended? On a related issue, is it true that a TOR-type region for which the high address is 0x0 is effectively always disabled?

**Caching of translation control attributes**
In the ARM architecture with which I am familiar, it is permitted that translation control fields such as mstatus.SUM and mstatus.MXR are cached in TLB entries. What this means is that the values of these fields are examined only when the TLB entry is created, breaking the tight dependency between fundamental status registers and instructions. The RISCV definition seems not to mention any such caching of translation attributes, meaning that the current value of mstatus is a required input to every instruction. Is this correct?

Thanks,

James.

aswaterman, 2018-02-20:
Re: MPRV: It is true that MPRV sequences need to be carefully handwritten.  The typical use case is to enable MPRV, do a load or store, then disable MPRV immediately afterwards.  There are plenty of other ways to accidentally grant excess privilege to untrusted software; these all amount to software bugs.

Re: PMPs: yes, you need to use a NAPOT PMP if you want to grant permissions to the entire address space in RV32.

In general, TOR PMPs only match an address if pmpaddr(i) <= addr < pmpaddr(i+1), so yeah, if pmpaddr(i+1) = 0, it's effectively disabled.

Re: caching MXR/SUM: you are right that these are effectively inputs to every load/store.  If you were to cache them in the TLB entries, you'd need to check and possibly invalidate the TLB entries on every MXR/SUM change.  Many system calls flip SUM at least twice, so that would probably perform very poorly.

JamesKenneyImperas, 2018-02-22:
Hello Andrew,

Regarding the TOR PMP constraint: note that it isn't just a question of granting permission to the entire address space: you can't use a TOR PMP to restrict access to _any_ range ending at the last address (e.g. the last 12Kb of memory). This might be an irritation for some people, but can be worked around as long as it is known about.

I think granting of excess privileges should be of more concern. The issue here is not software bugs but _malicious_ code. Here, I could envisage that a very small and difficult-to-detect modification to supposedly-trusted Machine mode source code would enable a cooperating malign User mode application to access all Supervisor mode memory, possibly without the Supervisor being aware at all.

aswaterman, 2018-11-10:
Yeah, I sympathize with that concern, but I think my point still stands.  There are myriad bits of M-mode where a very small, difficult-to-detect code modification could open a huge security hole.  I don't think the PMPs are an egregious offender.

closed 2018-11-10 by aswaterman