riscv/riscv-isa-manual #141: Feature bits (S-mode, U-mode, MMU/No-MMU)

michaeljclark, 2018-03-04:
It appears that the `misa` S bit is used informally to denote S-mode support, and likewise the `misa` U bit denotes U-mode support, however, the base ISA specification _22.8 Subset Naming Convention_ does not list the U bit, the privilege spec doesn't specify the meaning of the S and U bits and also there is currently no specified method to check for the presence of an MMU.

- https://github.com/michaeljclark/riscv-probe

```
$ spike bin/riscv64/probe-htif 
isa: rv64imafdcsu
[snip]
$ qemu-system-riscv64 -machine spike_v1.10 -nographic -kernel bin/riscv64/probe-htif 
isa: rv64imafdcsu
[snip]
```

I had a discussion with @palmer-dabbelt with respect to implementing QEMU support for cores that don't implement S-mode and/or U-mode. We agreed that silently ignoring writes to `mstatus.mpp` was the correct way to disallow mode switches to unsupported modes, and the misa bit is used to indicate whether the mode is supported. This is what I have implemented so that we can emulate the E-series cores. There is also the question of M-mode software detecting the presence of an MMU, which could either be handled with an illegal instruction trap when writing to `satp` or for writes to be ignored and for software to read back `satp` to see if the VM mode was accepted. Out of pragmatism, I made writes to `satp` trap when there is no MMU, however, this was done using an out-of-band feature bit, to enable or disable the MMU for emulating various CPU models. Currently, the other emulation features in QEMU are enabled or disabled based on precense of `misa` (or a `misa_mask` field which will hold the supported features when we implement support for writable `misa`).

It would be ideal if we have a non-device tree solution specified for MMU detection in low-level software i.e. boot loader code and tests that are shared between MMU and no-MMU systems. It seems the presence of U-mode, S-mode and MMU could be specified relatively easily:

Proposals

- update _22.8 Subset Naming Convention_ in to add S and U bit to Standard User Level extensions to indicate "Supervisor mode" and "User mode".
- update the privileged spec to state that the misa S bit indicates support for Supervisor mode.
- update the privileged spec to state that the misa U bit indicates support for User mode.
- add text that states a write of an unsupported mode to `mstatus.mpp` will be ignored and that software and that subsequent reads to `mstatus.mpp` will return the previous value.
- define MMU vs No-MMU behavior i.e. does a cssrr `satp` write fault fault, or are writes ignored and the register is hardwired to zero? or both?

Just thought i'd file an issue here so we can discuss.

aswaterman, 2018-03-05:
We've already specified that MPP ignores writes of unsupported values, for exactly this purpose: https://github.com/riscv/riscv-isa-manual/blob/master/src/machine.tex#L548

aswaterman, 2018-03-05:
Also, the misa.S/U behavior is formally specified: https://github.com/riscv/riscv-isa-manual/blob/master/src/machine.tex#L144

michaeljclark, 2018-03-05:
Oh good. I'll read the spec more closely.

- S-mode bit denotes presence of s* CSRs?
- s* CSRs trap if S-mode is not set?
- no-MMU with S-mode. Is satp hard-wired to zero or does it faults?

michaeljclark, 2018-03-05:
Okay Now I see them in "Table 3.2: Encoding of Extensions field in misa" of the privileged spec. I was looking at "22.8 Subset Naming Convention" in the Base ISA spec, which has Sdef, but not U.

aswaterman, 2018-03-05:
E is a replacement for I, so it has to come at the beginning, at the exclusion of I.

michaeljclark, 2018-03-05:
OK. I'll reorder E.

Rewriting my questions:

- misa S bit denotes the presence of s* CSRs?
- all s* CSRs must trap if misa S bit is not set?
- no-MMU with S bit. Is satp hard-wired to zero or does it fault?

aswaterman, 2018-03-05:
Strictly speaking, these are platform issues, not issues for the ISA spec itself.  But let's assume the following as sensible defaults:

- Yes
- Not exactly, because misa is not required to be implemented (i.e. it can be hardwired to 0).  However, I'd say that if misa is implemented, then yes, s* CSRs should trap is misa.S is clear.
- Hardwired to 0.

Does that sound OK?

michaeljclark, 2018-03-05:
This sounds good.

I'll make qemu drop writes to satp and return 0 when misa.S is set and no MMU is present.

closed 2018-11-21 by aswaterman