riscv/riscv-isa-manual #143: Writing reserved mode to mtvec/stvec

michaeljclark, 2018-03-05:
The specification does not state what happens when a _mode_ other than 0 or 1 is written to the lower 2-bits of mtvec/stvec.

- reserved modes imply future modes which implies there needs to be the specification for writes of unsupported modes
- If a reserved mode is written, is the whole write ignored? or is there a trap? one wouldn't want to try to set a mode where the base is set up for that specific mode but that mode is not supported
- Are vectored traps optional or mandatory? The spec reads such that they are mandatory, as it doesn't define any behavior if an unsupported mode is written. e.g. no vectored traps.

aswaterman, 2018-03-05:
MODE is WARL, which means it can be probed and won't cause a trap.  However, probing MODE may destroy the rest of the value in mtvec.

Vectored traps are optional.

michaeljclark, 2018-03-05:
I think we need to state that vectored traps are optional. It might already be stated and I have overlooked it. QEMU currently traps if the LSB of stvec or mtvec is set, which is apparently wrong. I'll fix it (it could have been me that broke it, I'd have to look through the commit history).

aswaterman, 2018-03-05:
It's already implicitly optional, because ```mtvec``` is allowed to be hardwired.  Regardless, which features are optional and which are required is a platform issue.

michaeljclark, 2018-03-05:
np. we are going to need to document this somewhere, in a forthcoming platform spec, as QEMU might like to emulate much more restricted implementations, hence my questions. implicit behavior is okay if you can reason about all of the rules concurrently. yes it's WARL, so understood. I've just fixed QEMU.

which reminds about S mode and U mode support. QEMU needs to mask mie/mip fields depending on whether S mode or U mode are implemented. in any case, we are getting closer to being able to emulate a more restricted implementation of the spec...

aswaterman, 2018-03-05:
Cool.  Yeah, mie/mip should force 0s into unsupported privilege modes' fields.

michaeljclark, 2018-03-05:
It would be nice to have all of these dimensions in a form that could be fed into a SAT solver and an implementation that could implement arbitrary restricted legal subsets of the spec.

aswaterman, 2018-03-05:
My guess is that there will be a relatively small number of platform profiles (say, 1-2 each for microcontrollers and application processors), so we probably won't need to enumerate large parts of the space.

closed 2018-11-21 by aswaterman